jkjung-avt/tensorrt_demos #333: Converting ssd_mobilenet_v2_mnasfpn to TensorRT

codethief, 2021-01-21:
Hi JK,

First of all, thank you so much for all the work you've been putting into making TensorRT accessible for people like me! I honestly do not know what I would do without your reference code. :)

Here goes my question: Have you had any luck converting ssd_mobilenet_v2_mnasfpn to TensorRT? (I am aware MnasFPN was optimized rather for CPUs than for GPUs but in my application I was seeing much better accuracy than with ssd_mobilenet_v2. Plus, I also need to run my network on a CPU sometimes.)

Anyway, I extended your `build_engine.py` file to attempt the conversion but it didn't work. Judging from the [logs](https://github.com/jkjung-avt/tensorrt_demos/files/5851942/tensorrt_mnasfpn_test.txt), it's the `MultiscaleGridAnchorGenerator` that's causing the trouble and I've searched the entire internet for any pointers – to no avail. Do you happen to have an idea on how I could proceed here?

For future reference, here's my code:

```
'ssd_mobilenet_v2_mnasfpn_custom': {
    'input_pb':   os.path.abspath(os.path.join(
                      DIR_NAME, 'ssd_mobilenet_v2_mnasfpn_custom.pb')),
    'tmp_uff':    os.path.abspath(os.path.join(
                      DIR_NAME, 'ssd_mobilenet_v2_mnasfpn_custom.uff')),
    'output_bin': os.path.abspath(os.path.join(
                      DIR_NAME, 'TRT_ssd_mobilenet_v2_custom.bin')),

    # Different shape than all other networks, so I created an input_dims 
    # parameter for each model
    'input_dims': (3, 320, 320), 

    # Again different than for all other models, so I moved it to a parameter, too. 
    # I obtained it using an adapted version of 
    # https://stackoverflow.com/questions/51742805/ssd-anchors-in-tensorflow-detection-api
    'feature_map_shapes': [40, 20, 10, 5],  

    'num_classes': 3,  # 2 classes + background class
    'min_size': 0.2,
    'max_size': 0.95,
    'input_order': [1, 0, 2],  # order of loc_data, conf_data, priorbox_data
},
```

jkjung-avt, 2021-01-23:
> [TensorRT] ERROR: UffParser: Validator error: MultiscaleGridAnchorGenerator/GridAnchorGenerator_3/Meshgrid_2/ExpandedShape_1/ones: Unsupported operation _Fill

TensorFlow Object Detection API code base is constantly evolving.  Based on this error, I'd think the "MultiscaleGridAnchorGenerator" in the "ssd_mobilenet_v2_mnasfpn" has changed quite a bit from older versions of the API.

By the way, the "MultiscaleGridAnchorGenerator" is used to be called "MultipleGridAnchorGenerator" in the older version of TensorFlow Object Detection API.  Refer to this line of source code:

https://github.com/jkjung-avt/tensorrt_demos/blob/451208d77af78d66f0fcbd70802b8bea1701eb5e/ssd/build_engine.py#L227

codethief, 2021-01-23:
Hi JK, thanks so much for your quick response! 

Are you sure MultiscaleGridAnchorGenerator and MultipleGridAnchorGenerator are referring to the same entity? As far as I can see, the TensorFlow Object Detection repository still contains both,

https://github.com/tensorflow/models/blob/master/research/object_detection/anchor_generators/multiscale_grid_anchor_generator.py
https://github.com/tensorflow/models/blob/master/research/object_detection/anchor_generators/multiple_grid_anchor_generator.py

and they look quite different. Or is it that MultiscaleGridAnchorGenerator evolved out of MultipleGridAnchorGenerator and the latter is only being kept in the Object Detection repo for legacy/compatibility purposes?

In any case, I'll try to adapt `build_engine.py` as you suggested and will get back to you as soon as possible!

Again, thank you so much!

jkjung-avt, 2021-01-23:
> Or is it that MultiscaleGridAnchorGenerator evolved out of MultipleGridAnchorGenerator and the latter is only being kept in the Object Detection repo for legacy/compatibility purposes?

I think this is likely the case.

codethief, 2021-01-24:
The naive adaptation of `build_engine.py` didn't do the trick, unfortunately:

```
$ python3 build_engine.py ssd_mobilenet_v2_mnasfpn_custom
2021-01-24 17:28:44.711358: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.10.2
WARNING:tensorflow:Deprecation warnings have been disabled. Set TF_ENABLE_DEPRECATION_WARNINGS=1 to re-enable them.
NOTE: UFF has been tested with TensorFlow 1.15.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.9
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 320
      }
      dim {
        size: 320
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:226: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: GridAnchor_TRT yet.
Converting MultiscaleGridAnchorGenerator as custom op: GridAnchor_TRT
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:143] Marking ['NMS'] as outputs
No. nodes: 1185
UFF Output written to /home/user/tensorrt_demos/ssd/ssd_mobilenet_v2_mnasfpn_custom.uff
UFF Text Output written to /home/user/tensorrt_demos/ssd/ssd_mobilenet_v2_mnasfpn_custom.pbtxt
#assertiongridAnchorPlugin.cpp,445
Aborted (core dumped)
```

I should mention that the script was hanging at `#assertiongridAnchorPlugin.cpp,445` for a couple minutes. Do you happen to know how I could debug this and find out where exactly things went wrong? Where does the core get dumped?

For definiteness, below is my modified `build_engine.py`. As discussed, I mapped `MultiscaleGridAnchorGenerator` to the `GridAnchor_TRT` op, similarly to `MultipleGridAnchorGenerator`:

```python

"""build_engine.py

This script converts a SSD model (pb) to UFF and subsequently builds
the TensorRT engine.
[... 305 lines cut ...]
        model,
        spec)
    _ = uff.from_tensorflow(
        dynamic_graph.as_graph_def(),
        output_nodes=['NMS'],
        output_filename=spec['tmp_uff'],
        text=True,
        debug_mode=DEBUG_UFF)
    with trt.Builder(TRT_LOGGER) as builder, builder.create_network() as network, trt.UffParser() as parser:
        builder.max_workspace_size = 1 << 28
        builder.max_batch_size = 1
        builder.fp16_mode = True

        parser.register_input('Input', spec['input_dims'])
        parser.register_output('MarkOutput_0')
        parser.parse(spec['tmp_uff'], network)
        engine = builder.build_cuda_engine(network)

        buf = engine.serialize()
        with open(spec['output_bin'], 'wb') as f:
            f.write(buf)

if __name__ == '__main__':
    main()
```

codethief, 2021-01-24:
Never mind my previous comment, I had made a mistake and it does seem to work to some degree – I'm now getting a much "nicer" error (see below).

My mistake was that, if I understand the [grid anchor plugin's README](https://github.com/NVIDIA/TensorRT/blob/master/plugin/gridAnchorPlugin/README.md) correctly, the `numLayers` parameter must be the length of the given `featureMapShapes` list. I've adjusted my code accordingly:

```python
    PriorBox_MultiscaleGridAnchorGenerator = gs.create_plugin_node(
        name='MultiscaleGridAnchorGenerator',
        op='GridAnchor_TRT',
        minSize=minSize,  # was 0.2
        maxSize=maxSize,  # was 0.95
        aspectRatios=[1.0, 2.0, 0.5, 3.0, 0.33],
        variance=[0.1, 0.1, 0.2, 0.2],
        featureMapShapes=spec['feature_map_shapes'],
        numLayers=len(spec['feature_map_shapes']),
    )
```

(Once again, I'm attaching [the full code](https://github.com/jkjung-avt/tensorrt_demos/files/5862834/build_engine.py.txt) for definiteness.)

The script now fails at the following step:

```
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/weights[Op: Const].
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/weights -> [1,1,256,208]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/weights
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/Conv2D[Op: Conv]. Inputs: FeatureExtractor/MobilenetV2/expanded_conv_16/project/BatchNorm/FusedBatchNormV3, FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/weights
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/Conv2D -> [208,10,10]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/Conv2D
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/gamma[Op: Const].
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/gamma -> [208]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/gamma
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/beta[Op: Const].
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/beta -> [208]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/beta
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_mean[Op: Const].
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_mean -> [208]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_mean
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_variance[Op: Const].
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_variance -> [208]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_variance
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/FusedBatchNormV3[Op: BatchNorm]. Inputs: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/Conv2D, FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/gamma, FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/beta, FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_mean, FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/moving_variance
[TensorRT] VERBOSE: UFFParser: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/FusedBatchNormV3 -> [208,10,10]
[TensorRT] VERBOSE: UFFParser: Applying order forwarding to: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/FusedBatchNormV3
[TensorRT] VERBOSE: UFFParser: Parsing FeatureExtractor/MnasFPN/cell_0/block_0/input_1/nearest_neighbor_upsampling/stack[Op: Stack]. Inputs: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/FusedBatchNormV3, FeatureExtractor/MnasFPN/cell_0/block_0/input_1/Conv1x1/BatchNorm/FusedBatchNormV3
[TensorRT] ERROR: UffParser: Parser error: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/nearest_neighbor_upsampling/stack: stack only support for Constants nodes as input for now
[TensorRT] ERROR: Network must have at least one output
[TensorRT] ERROR: Network validation failed.
Traceback (most recent call last):
  File "build_engine.py", line 344, in <module>
    main()
  File "build_engine.py", line 338, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'
```

([Full logs](https://github.com/jkjung-avt/tensorrt_demos/files/5862839/logs.txt))

Unfortunately, I seem to have reached a dead end if interpret [this answer on the Nvidia forums](https://forums.developer.nvidia.com/t/tensorrt-stack-stack-only-support-for-constants-nodes-as-input-for-now/109172/2) correctly. :(

jkjung-avt, 2021-01-25:
> [TensorRT] ERROR: UffParser: Parser error: FeatureExtractor/MnasFPN/cell_0/block_0/input_1/nearest_neighbor_upsampling/stack: stack only support for Constants nodes as input for now

The "Stack" op could probably be replaced by a couple of "Reshape"s and a "Concat".  If you are able to use graphsurgeon to modify the graph, it might be worth a try.

Reference: https://stackoverflow.com/questions/41534593/why-would-i-ever-use-tf-concat-instead-of-tf-stack

codethief, 2021-02-03:
Oops, forgot to answer. I'll be giving ONNX a shot in the coming days. If that fails, I'll try to replace the Stack op. Thanks so much for your help so far! I'll report back!

closed 2021-02-17 by jkjung-avt