jkjung-avt/tensorrt_demos #334: Problems converting yolov4-tiny-3layer models to TRT

aeozyalcin, 2021-01-22:
Hello JK,

I am trying to convert a custom-trained [yolov4-tiny-3layer](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4-tiny-3l.cfg) model to tensorRT to do accelerated inference on the Jetson Nano. However I haven't successful, I was curious if this was possible at all using your library.

Here is what I tried:

- Start with "yolov4-tiny-3l-512x512.weights" generated by Darknet, conversion to onnx successful, but when going from onnx to trt, I get the following error. I am assuming the converter automatically assumes it's a 2layer network because it says "tiny" in the file name:
```aeozyalcin@aeozyalcin-nano:~/tensorrt_demos/yolo$ python3 onnx_to_tensorrt.py -m yolov4-tiny-3l-512x512 -c 1
Loading the ONNX file...
Adding yolo_layer plugins...
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 195, in <module>
    main()
  File "onnx_to_tensorrt.py", line 184, in main
    args.model, args.category_num, args.int8, args.dla_core, args.verbose)
  File "onnx_to_tensorrt.py", line 113, in build_engine
    network, model_name, category_num, TRT_LOGGER)
  File "/home/aeozyalcin/tensorrt_demos/yolo/plugins.py", line 131, in add_yolo_plugins
    (len(anchors), len(yolo_whs)))
ValueError: bad number of yolo layers: 3 vs. 2
```

- So I got rid of the "tiny" in the filename, and tried to convert "yolov4-3l-512x512.weights" to TRT. Conversion to onnx was successful, but conversion to trt failed with a segmentation fault:
```
...
[TensorRT] VERBOSE: ModelImporter.cpp:179: 043_convolutional_lrelu [LeakyRelu] outputs: [043_convolutional_lrelu -> (1, 128, 64, 64)],
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: 044_convolutional [Conv]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 043_convolutional_lrelu
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 044_convolutional_conv_weights
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 044_convolutional_conv_bias
[TensorRT] VERBOSE: ModelImporter.cpp:125: 044_convolutional [Conv] inputs: [043_convolutional_lrelu -> (1, 128, 64, 64)], [044_convolutional_conv_weights -> (18, 128, 1, 1)], [044_convolutional_conv_bias -> (18)],
[TensorRT] VERBOSE: builtin_op_importers.cpp:450: Convolution input dimensions: (1, 128, 64, 64)
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: 044_convolutional for ONNX node: 044_convolutional
[TensorRT] VERBOSE: builtin_op_importers.cpp:533: Using kernel: (1, 1), strides: (1, 1), prepadding: (0, 0), postpadding: (0, 0), dilations: (1, 1), numOutputs: 18
[TensorRT] VERBOSE: builtin_op_importers.cpp:534: Convolution output dimensions: (1, 18, 64, 64)
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 044_convolutional for ONNX tensor: 044_convolutional
[TensorRT] VERBOSE: ModelImporter.cpp:179: 044_convolutional [Conv] outputs: [044_convolutional -> (1, 18, 64, 64)],
Segmentation fault (core dumped)
```

I am having some trouble eliminating false positives with the traditional 2 layer yolov4-tiny model, so wanted to increase network complexity by trying the yolov4-tiny-3l model, to trade speed for accuracy. But have not been able to convert to trt unfortunately.

I have attached my Darknet weights and cfg files on here, if you wanted to give it a try. Thank you!

[yolov4-tiny-3l-512x512.zip](https://github.com/jkjung-avt/tensorrt_demos/files/5857777/yolov4-tiny-3l-512x512.zip)

jkjung-avt, 2021-01-23:
You need to figure out the output layer numbers and modify the source code accordingly.  Please refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/268#issuecomment-728915854

aeozyalcin, 2021-01-23:
Thanks @jkjung-avt .

For others interested in converting yolov4-tiny-3l models into tensorRT, here is what I had to change to get it to work:

yolo_to_onnx.py:
```
...
    elif 'yolov4' in args.model:
        if 'tiny-3l' in args.model:
            output_tensor_dims['030_convolutional'] = [c, h // 32, w // 32]
            output_tensor_dims['037_convolutional'] = [c, h // 16, w // 16]
            output_tensor_dims['044_convolutional'] = [c, h // 8, w // 8]
        elif 'yolov4x-mish' in args.model:
            output_tensor_dims['168_convolutional_lgx'] = [c, h //  8, w //  8]
            output_tensor_dims['185_convolutional_lgx'] = [c, h // 16, w // 16]
            output_tensor_dims['202_convolutional_lgx'] = [c, h // 32, w // 32]
        elif 'yolov4-csp' in args.model:
            output_tensor_dims['144_convolutional_lgx'] = [c, h //  8, w //  8]
            output_tensor_dims['159_convolutional_lgx'] = [c, h // 16, w // 16]
            output_tensor_dims['174_convolutional_lgx'] = [c, h // 32, w // 32]
        elif 'tiny' in args.model:
            output_tensor_dims['030_convolutional'] = [c, h // 32, w // 32]
            output_tensor_dims['037_convolutional'] = [c, h // 16, w // 16]
        else:
            output_tensor_dims['139_convolutional'] = [c, h //  8, w //  8]
            output_tensor_dims['150_convolutional'] = [c, h // 16, w // 16]
            output_tensor_dims['161_convolutional'] = [c, h // 32, w // 32]
...
```

plugins.py:
```
def get_yolo_whs(model_name, w, h):
    """Get yolo_width and yolo_height for all yolo layers in the model."""
    if 'yolov3' in model_name:
        if 'tiny' in model_name:
            return [[w // 32, h // 32], [w // 16, h // 16]]
        else:
            return [[w // 32, h // 32], [w // 16, h // 16], [w // 8, h // 8]]
    elif 'yolov4' in model_name:
        if 'tiny-3l' in model_name:
            return [[w // 32, h // 32], [w // 16, h // 16], [w // 8, h // 8]]
        elif 'tiny' in model_name:
            return [[w // 32, h // 32], [w // 16, h // 16]]
        else:
            return [[w // 8, h // 8], [w // 16, h // 16], [w // 32, h // 32]]
    else:
        raise ValueError('ERROR: unknown model (%s)!' % args.model)
```

yolo_with_plugins.py:
```
def get_yolo_grid_sizes(model_name, h, w):
    """Get grid sizes (w*h) for all yolo layers in the model."""
    if 'yolov3' in model_name:
        if 'tiny' in model_name:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16)]
        else:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16), (h // 8) * (w // 8)]
    elif 'yolov4' in model_name:
        if 'tiny-3l' in model_name:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16), (h // 8) * (w // 8)]
        elif 'tiny' in model_name:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16)]
        else:
            return [(h // 8) * (w // 8), (h // 16) * (w // 16), (h // 32) * (w // 32)]
    else:
        raise ValueError('ERROR: unknown model (%s)!' % args.model)
```

Yields 33MB .trt files, and was able to do inference correctly. Make sure "yolov4-tiny-3l" is in the model name. JK you can feel free to verify the code and integrate into your code if you would like to.

closed 2021-01-23 by aeozyalcin

muhammadAgfian96, 2021-02-18:
Thanks. I achived 8-9 FPS with yolov4-tiny-3l-800