jkjung-avt/tensorrt_demos #335: Jetson xavier nx ERROR!

amphancm, 2021-01-23:
Hi JK,
I've use on Jetson xavier ,Jetpack SDK 4.4 .
got error. pls help. Thank you.

**jetson@xavier**:~/tensorrt_demos$ python3 trt_ssd_async.py --model ssd_mobilenet_v1_coco --video_looping --video /home/jetson/Videos/traffic_960x540.mp4

TrtThread: loading the TRT SSD engine...
**[TensorRT] ERROR: INVALID_CONFIG: The engine plan file is generated on an incompatible device, expecting compute 7.2 got compute 5.3, please rebuild.**
[TensorRT] ERROR: engine.cpp (1546) - Serialization Error in deserialize: 0 (Core engine deserialization failure)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/jetson/tensorrt_demos/utils/ssd.py", line 88, in __init__
    self.context = self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "trt_ssd_async.py", line 97, in run
    self.trt_ssd = TrtSSD(self.model, INPUT_HW)
  File "/home/jetson/tensorrt_demos/utils/ssd.py", line 92, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources

Exception ignored in: <bound method TrtSSD.__del__ of <utils.ssd.TrtSSD object at 0x7f64db2cf8>>
Traceback (most recent call last):
  File "/home/jetson/tensorrt_demos/utils/ssd.py", line 99, in __del__
    del self.cuda_outputs
AttributeError: cuda_outputs
-------------------------------------------------------------------
PyCUDA ERROR: The context stack was not empty upon module cleanup.
-------------------------------------------------------------------
A context was still active when the context stack was being
cleaned up. At this point in our execution, CUDA may already
have been deinitialized, so there is no way we can finish
cleanly. The program will be aborted now.
Use Context.pop() to avoid this problem.
-------------------------------------------------------------------

jkjung-avt, 2021-01-24:
> [TensorRT] ERROR: INVALID_CONFIG: The engine plan file is generated on an incompatible device, expecting compute 7.2 got compute 5.3, please rebuild.

You need to rebuild the TensorRT engine on the Jetson Xavier.

amphancm, 2021-01-24:
> > [TensorRT] ERROR: INVALID_CONFIG: The engine plan file is generated on an incompatible device, expecting compute 7.2 got compute 5.3, please rebuild.
> 
> You need to rebuild the TensorRT engine on the Jetson Xavier.

Thank you for very fast response. I'll try rebuild that.

jkjung-avt, 2021-02-03:
Has this issue been fixed?

jkjung-avt, 2021-02-06:
Closing due to no response.

closed 2021-02-06 by jkjung-avt