jkjung-avt/tensorrt_demos #337: Category number

dilshan-n-wickramarachchi, 2021-01-25:
Hi, 
I'm trying to use yolov4-tiny custom trained model, Please explain me about the category number. 
I have 6 classes but when creating the engine i get the error bad category number (6)

dilshan-n-wickramarachchi, 2021-01-25:
[yolov4_tiny.zip](https://github.com/jkjung-avt/tensorrt_demos/files/5866969/yolov4_tiny.zip)

jkjung-avt, 2021-01-26:
Did you specify "--category_num" or "-c" in the command?  For example,

```
$ python3 yolo_to_onnx.py -c 6 -m custom-yolov4-tiny-416x416
```

jkjung-avt, 2021-02-03:
Closing due to no response.

closed 2021-02-03 by jkjung-avt

zoombinis, 2021-05-02:
@jkjung-avt I'm curious about the OP's first question, is there deeper clarification available on how to configure the number of categories? For instance, I'm curious to know:

- If I only want "person" and "car" detected, would that mean I need to set `category_num` to `3` (per [COCO_CLASSES_LIST](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolo_classes.py#L6))?  
- Is there a way to choose only "person" and "car" (not "bicycle")?

> I'm trying to use yolov4-tiny custom trained model, Please explain me about the category number.

jkjung-avt, 2021-05-03:
"category_num" in my code refers to "how many object categories the YOLO model has been trained to recognize".  For YOLO models trained with MS COCO dataset, this number should always be 80.

If you only want to keep a subset of object categories, you could filter class id's after you get detection results from the model.