jkjung-avt/tensorrt_demos #338: About Custom Plugins

Darshcg, 2021-01-26:
Hi @jkjung-avt,

Thanks for your Project. 
I just wanted to know where you have used Custom Plugins(where the Plugins have been called) and for what purpose you are using the Custom Plugins?

Thank you,
Darshan

jkjung-avt, 2021-01-27:
I use "yolo_layer" plugin in the TensorRT yolov3/yolov4 implementation.  Source code of the plugin could be found in [plugins](https://github.com/jkjung-avt/tensorrt_demos/tree/master/plugins).

The "yolo_layer" plugins are inserted into the TensorRT yolov3/yolov4 network in "onnx_to_tensorrt.py".  They implement computations in the original darknet "yolo" layers which could be efficiently parallelized by CUDA kernel function.

https://github.com/jkjung-avt/tensorrt_demos/blob/793d7ae9c74b51007296bb63cdd1bc6bd0e04e8e/yolo/onnx_to_tensorrt.py#L112-L113

Darshcg, 2021-01-27:
Thank you for your response @jkjung-avt. 

Actually, I have a custom models built from Pytorch, I have converted .pt to .onnx, but there are two Ops which are not supported in TensorRT. So, I am thinking of adding the custom plugins of those two unsupported Ops in TensorRT.
As per my Knowledge, there are two ways to implement the custom Layers(through C++ and Python Network Defination API).
Having said that, As I have a ONNX model, what is the better approach? which one to go for? what are the challenges in adding the custom Plugins?

Thanks in advance,
Darshan

jkjung-avt, 2021-01-27:
These questions don't really relate to code in this repo.  I won't be able to provide much help going forward.

Please refer to:

* https://developer.nvidia.com/blog/estimating-depth-beyond-2d-using-custom-layers-on-tensorrt-and-onnx-models/
* https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#extending

Darshcg, 2021-01-27:
Thank you very much @jkjung-avt.
I understand. I will refer to the above references sent by you.

Darshcg, 2021-01-27:
Hi @jkjung-avt,

I was understanding about the adding the Custom layers to the TensorRT Network by looking at your Implementation(in "yolo_layer.cu" and "yolo_layer.h"), having said that, I did'nt get where you are adding the logic of that Plugin?? Can you please help me in understanding that?

jkjung-avt, 2021-01-27:
I have already highlighted the code piece in https://github.com/jkjung-avt/tensorrt_demos/issues/338#issuecomment-767950754.

Darshcg, 2021-01-27:
Hi @jkjung-avt,

Thank you for your reply. 
You said that:"The "yolo_layer" plugins are inserted into the TensorRT yolov3/yolov4 network in "onnx_to_tensorrt.py". They implement computations in the original darknet "yolo" layers which could be efficiently parallelized by **CUDA kernel function**"

But where is the **CUDA kernel function** implemented in "yolo_layer.cu" and "yolo_layer.h"?

Thanks

jkjung-avt, 2021-01-27:
> But where is the CUDA kernel function implemented in "yolo_layer.cu" and "yolo_layer.h"?

https://github.com/jkjung-avt/tensorrt_demos/blob/451208d77af78d66f0fcbd70802b8bea1701eb5e/plugins/yolo_layer.cu#L203

and

https://github.com/jkjung-avt/tensorrt_demos/blob/451208d77af78d66f0fcbd70802b8bea1701eb5e/plugins/yolo_layer.cu#L261

closed 2021-01-30 by jkjung-avt