jkjung-avt/tensorrt_demos #339: illeagel instruction core dumped

basitraza1214, 2021-02-03:
I have been using this source code to run models successfully but now whenever I am trying to load a model it outputs an error
Illegal-instruction (core dumped)
Can somebody please highlight the issue here?

jkjung-avt, 2021-02-03:
This description of issue is too rough...

Maybe you should use [gdb](https://www.thegeekstuff.com/2014/01/gdb-backtrace/) to find where the "illegal instruction" error occurs, i.e. in which library of which function/operation.

closed 2021-02-11 by jkjung-avt

basitraza1214, 2021-02-18:
Hey
Sorry for not replying, I was out of town and could not check my mail but thankfully I was able to solve the issue here.
Thank you so very much for your help.

Regards,

Syed Basit Raza Bukhari
Research Consultant,
Cruise Aerospace, Islamabad, Pakistan
For Avionics Engineering Department
College of Aeronautical Engineering (CAE)
National University of Science and Technology (NUST)
Risalpur, Khyber Pakhtunkhwa Pakistan.
+92(923)014096964

Closed #339<https://github.com/jkjung-avt/tensorrt_demos/issues/339>.

MayurHulke, 2021-02-22:
How did you solved the issue ?

basitraza1214, 2021-02-22:
Well, it turns out that I was making some silly mistakes.
What I did was to flash jetpack 4.4 and use the built-in OpenCV.
I installed pip3 and then directly followed the tutorial. One thing that I did was to make and mount a swap file and installed protobuf from getting started with the jetson nano blog (if I am correct about the blog title) and everything was running perfectly again.

dlavrantonis, 2022-12-22:
for anyone facing the same issue, see the below info:
For me the problem happened when executing import onnx
After googling it, it turned out that the fix was to downgrade numpy:  pip3 install numpy==1.19.4
details on the issue  [here](https://github.com/onnx/onnx/issues/3268)

jkjung-avt, 2022-12-23:
@dlavrantonis   Thanks for sharing.