jkjung-avt/tensorrt_demos #340: I have a question about using yolo (transfer learning)

ingbeeedd, 2021-02-03:
Hello, I have a question regarding `yolo transfer learning`. 
In Demo 4, I can see that your script about the weight, cfg file directly from the yolo official site through the download_yolo script. 
If I take a model of `yolo transfer learning` somewhere else, is this also applicable about tensorrt? 
If you can answer "Yes" it, can you introduce it if there is a reference site that can be customized? thank you

jkjung-avt, 2021-02-03:
In step 6 of [Demo 5: YOLOv4](https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4), I demonstrated converting a custom trained "yolov4-crowdhuman" model (with only 2 classes of objects) to TensorRT engine.

As also stated in the README, you could refer to my [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/) blog post for more information.

ingbeeedd, 2021-02-04:
Ah! The class for Infinite War is one that is not in the existing dataset(COCO or VOC). Thank you. I will refer to it and proceed with the development.

closed 2021-02-04 by ingbeeedd