jkjung-avt/tensorrt_demos #346: [Question] Best model for object detection

leviethung2103, 2021-02-17:
Hi author, 

First of all, thank you so much for creating this repository. It helps me a lot when I am getting to know the Jetson Nano and TensorRT. 

I've tested the SSD Model you provided. It achieved a very good FPS ~  24-26 FPS. However, I saw that the accuracy is not good in several cases. The model outputs a lot of False Positive detection. I am wondering that are there any alternatives for object detection problem which is better than SSD? 

Thank you in advance.

jkjung-avt, 2021-02-17:
If you are looking to do real-time near 30FPS object detection on Jetson Nano, I think **yolov4-tiny** is the best choice right now.

leviethung2103, 2021-02-17:
> If you are looking to do real-time near 30FPS object detection on Jetson Nano, I think **yolov4-tiny** is the best choice right now.

Thank you for your suggestion. I will try the Yolov4-tiny first.

closed 2021-02-17 by leviethung2103