jkjung-avt/tensorrt_demos #348: Generating Alarms upon detection

basitraza1214, 2021-02-18:
Hi jkjung,
 Thank you very much for this wonderful work here, you are a big help.
I am currently working on a project and was hoping that if you could point me in the right direction. What I am trying to achieve is to do an object detection using a custom trained model, which I was able to achieve thanks to you and stream the live feed to a remote pc on the network, secondly, I need to generate an alarm when a detection takes place (please help me in this).
Now I am using yolov4 and using your trt_yolo_mjpeg.py file as source. Now one course of action in my mind was to somehow create a JSON file of every detection and then use a separate code to continually check through the JSON file for any new entry and generate an alarm in that case.
I would be grateful to you for your help.
Currently, I am working on Jetson Nano with jetpack 4.4.

jkjung-avt, 2021-02-19:
The most relevant line of code is:

https://github.com/jkjung-avt/tensorrt_demos/blob/32ce1c39ae74c9cf0b93e491ecefbe3453a855c8/trt_yolo_mjpeg.py#L62

The detection outputs are bounding box coordinates (x1, y1, x2, y2), confidence scores and class IDs.  You could use those to create your own JSON file or generate alarms/events.

basitraza1214, 2021-02-20:
thank you so much, I will try this and get back to you.
once again thank you so much

closed 2021-02-20 by jkjung-avt