jkjung-avt/tensorrt_demos #349: onnx失败

1124676457, 2021-02-19:
import onnx 出现Illegal instruction (core dumped)，版本也是对的，1.4.1，就是报错，protbuf也装了

jkjung-avt, 2021-02-19:
I haven't seen this issue before.  Maybe you could reference the following post:

https://github.com/onnx/onnx/issues/3268#issuecomment-777629332

1124676457, 2021-02-19:
好的，已经解决了，多谢

1124676457, 2021-02-19:
我还有个疑问，我在jetson nano上测试了一下，只有6帧左右，没有文档里说的那样快，是有什么问题吗

jkjung-avt, 2021-02-19:
If you set up the Jetson Nano software environment exactly as how I did it, you should get similar FPS numbers.

* [JetPack-4.5](https://jkjung-avt.github.io/jetpack-4.5/)
* [JetPack-4.4 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.4/)
* [JetPack-4.3 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.3/)

1124676457, 2021-02-20:
解决了，是计算方式有问题，多谢

closed 2021-02-20 by jkjung-avt

jkjung-avt, 2022-05-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462