jkjung-avt/tensorrt_demos #350: ValueError: bad number of network outputs: 1 vs. 3

muhammadAgfian96, 2021-02-19:
Hi Thanks for your contribution in this repo.

i've train yolov4-csp with the size 256,256 and class=1, and export .pt --> .onnx in colab. 
I tried to convert onnx (from colab) to Trt using onnx_ to_tensorrt.py but got this error.
```
...
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Reshape_528 for ONNX node: Reshape_528
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 815 for ONNX tensor: 815
[TensorRT] VERBOSE: ModelImporter.cpp:179: Reshape_528 [Reshape] outputs: [815 -> (1, 768, 6)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: Conv_529 [Conv]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 759
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: model.31.m.2.weight
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: model.31.m.2.bias
[TensorRT] VERBOSE: ModelImporter.cpp:125: Conv_529 [Conv] inputs: [759 -> (1, 1024, 8, 8)], [model.31.m.2.weight -> (18, 1024, 1, 1)], [model.31.m.2.bias -> (18)], 
[TensorRT] VERBOSE: builtin_op_importers.cpp:450: Convolution input dimensions: (1, 1024, 8, 8)
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Conv_529 for ONNX node: Conv_529
[TensorRT] VERBOSE: builtin_op_importers.cpp:533: Using kernel: (1, 1), strides: (1, 1), prepadding: (0, 0), postpadding: (0, 0), dilations: (1, 1), numOutputs: 18
[TensorRT] VERBOSE: builtin_op_importers.cpp:534: Convolution output dimensions: (1, 18, 8, 8)
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 816 for ONNX tensor: 816
[TensorRT] VERBOSE: ModelImporter.cpp:179: Conv_529 [Conv] outputs: [816 -> (1, 18, 8, 8)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: Reshape_543 [Reshape]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 816
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 833
[TensorRT] VERBOSE: ModelImporter.cpp:125: Reshape_543 [Reshape] inputs: [816 -> (1, 18, 8, 8)], [833 -> (5)], 
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Reshape_543 for ONNX node: Reshape_543
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 834 for ONNX tensor: 834
[TensorRT] VERBOSE: ModelImporter.cpp:179: Reshape_543 [Reshape] outputs: [834 -> (1, 3, 6, 8, 8)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: Transpose_544 [Transpose]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 834
[TensorRT] VERBOSE: ModelImporter.cpp:125: Transpose_544 [Transpose] inputs: [834 -> (1, 3, 6, 8, 8)], 
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Transpose_544 for ONNX node: Transpose_544
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 835 for ONNX tensor: 835
[TensorRT] VERBOSE: ModelImporter.cpp:179: Transpose_544 [Transpose] outputs: [835 -> (1, 3, 8, 8, 6)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: Sigmoid_545 [Sigmoid]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 835
[TensorRT] VERBOSE: ModelImporter.cpp:125: Sigmoid_545 [Sigmoid] inputs: [835 -> (1, 3, 8, 8, 6)], 
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Sigmoid_545 for ONNX node: Sigmoid_545
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 836 for ONNX tensor: 836
[TensorRT] VERBOSE: ModelImporter.cpp:179: Sigmoid_545 [Sigmoid] outputs: [836 -> (1, 3, 8, 8, 6)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: Reshape_548 [Reshape]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 836
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 842
[TensorRT] VERBOSE: ModelImporter.cpp:125: Reshape_548 [Reshape] inputs: [836 -> (1, 3, 8, 8, 6)], [842 -> (3)], 
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Reshape_548 for ONNX node: Reshape_548
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 843 for ONNX tensor: 843
[TensorRT] VERBOSE: ModelImporter.cpp:179: Reshape_548 [Reshape] outputs: [843 -> (1, 192, 6)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: Concat_549 [Concat]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 787
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 815
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 843
[TensorRT] VERBOSE: ModelImporter.cpp:125: Concat_549 [Concat] inputs: [787 -> (1, 3072, 6)], [815 -> (1, 768, 6)], [843 -> (1, 192, 6)], 
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: Concat_549 for ONNX node: Concat_549
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: output_1 for ONNX tensor: output
[TensorRT] VERBOSE: ModelImporter.cpp:179: Concat_549 [Concat] outputs: [output -> (1, 4032, 6)], 
[TensorRT] VERBOSE: ModelImporter.cpp:507: Marking output_1 as output: output
Adding yolo_layer plugins...
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 195, in <module>
    main()
  File "onnx_to_tensorrt.py", line 184, in main
    args.model, args.category_num, args.int8, args.dla_core, args.verbose)
  File "onnx_to_tensorrt.py", line 113, in build_engine
    network, model_name, category_num, TRT_LOGGER)
  File "/home/*****/project/tensorrt_demos/yolo/plugins.py", line 134, in add_yolo_plugins
    (network.num_outputs, len(anchors)))
ValueError: bad number of network outputs: 1 vs. 3
```

what should we check/fix?
Thanks.

muhammadAgfian96, 2021-02-20:
hmm... it seems we should convert onnx in our machine.

jkjung-avt, 2021-02-20:
You might need to modify the source code for your own custom trained model.  Refer to, say: https://github.com/jkjung-avt/tensorrt_demos/issues/334#issuecomment-765809470

closed 2021-02-28 by jkjung-avt