jkjung-avt/tensorrt_demos #351: trt_mtcnn.py crashes after awhile when running on RTSP

Bvgo, 2021-02-20:
WARN:1] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (488) isPipelinePlaying OpenCV | GStreamer warning: unable to query pipeline state
[ WARN:1] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module rtspsrc0 reported: Could not read from resource.
0 face(s) found

ran on Jetpack 4.5

jkjung-avt, 2021-02-20:
The problem could be either issue of the RTSP stream or incompatibility between GStreamer and the RTSP source.  I think I won't be able to solve or improve it at the moment.

leviethung2103, 2021-02-20:
I think you can double check the RTSP stream is correct and playable through the VLC player or you can turn off the usage of GSTREAMER flag inside the `camera.py` file.

Bvgo, 2021-02-26:
The problem was from the VLC timing out. I've resolved that using V4L2Rtspserver, which is a better alternative for our use case. It's works awesome now.

Moving forward! How can we initiate image capture and a short video recording? 

Thanks

jkjung-avt, 2021-02-28:
Video recording could be achieved by cv2.VideoWriter().  You might refer to [trt_yolo_cv.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_yolo_cv.py) as an example.

Bvgo, 2021-02-28:
Forgive me for being naive but How do I do this on the script, I keep getting one this wrong or the next. Kindly help

jkjung-avt, 2021-03-01:
You need to modify "trt_mtcnn.py" by yourself:

* Create a VideoWriter object: `out = cv2.VideoWriter(...)`
* Call the write() method of the VideoWriter for each frame of output: `out.write(frame)`
* Release the VideoWriter object when done: `out.release()`

jkjung-avt, 2021-03-01:
I'm closing this issue now since the video recording question has also been posted on NVIDIA Developer forum: https://forums.developer.nvidia.com/t/how-to-capture-images-and-record-short-videos-using-kafka/169569.

closed 2021-03-01 by jkjung-avt