jkjung-avt/tensorrt_demos #352: Linking the Custom plugins for the Conversion(while converting from ONNX to TRT Model)

Darshcg, 2021-02-20:
Hi, @jkjung-avt,

Thanks for your wonderful work.

I have Implemented two custom Plugins for two unsupported Ops. I want to link those custom Plugins for the conversion(while converting from ONNX to TRT Model). Can you please assist me in doing the same? Would be a great help.

Thanks and Regards,
Darshan

Darshcg, 2021-02-21:
Hi @jkjung-avt,

Actually, I am facing difficulty during converting the ONNX model to TRT with a Custom plugin. As I am doing this for the first time, your help would be highly appreciable. It would be a great help if you can help me in the same.

Waiting for your reply.

Thanks,
Darshan

Darshcg, 2021-02-21:
Hi @jkjung-avt ,

I got to know about the same. It is about including the .so file while conversion.
Example: ./trtexec --onnx=model.onnx --explicitBatch --shapes=input1:1x1x1x1 --plugins=layer.so

But when I run this I am getting an issue as: **Einsum_194: PluginV2Layer must be V2DynamicExt when there are runtime input dimensions**

The image of the error has been attached below:
[image]

Can you please help me to resolve this error? Where Exactly I am going wrong? what's the solution for this?

Thanks and Regards,
Darshan C G

jkjung-avt, 2021-02-25:
Closing due to duplication of this issue (https://github.com/NVIDIA/TensorRT/issues/1073) and not being related to code in this repo...

closed 2021-02-25 by jkjung-avt