jkjung-avt/tensorrt_demos #354: Yolo Dynamic engine

IsraelLencina, 2021-02-23:
Hi, i'm using your tensorRT generated from yolo_to_onnx -> onnx_to_yolo, the problem come when i try to use your engine from Triton inference server in mode dynamic batch size. 
I have located the issue in onnx_yo_tensorrt.py, the line:

```
profile.set_shape(
                '000_net',                                                             # input tensor name
                (MAX_BATCH_SIZE, 3, net_h, net_w),               # min shape
                (MAX_BATCH_SIZE, 3, net_h, net_w),               # opt shape
                (MAX_BATCH_SIZE, 3, net_h, net_w))               # max shape
```
But when i run this code TensorRT say me that the onnx is static in (32,3,416,416).

jkjung-avt, 2021-02-23:
Check if setting "batch=1" ([reference](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4.cfg#L2)) in the cfg file and redoing "yolo_to_onnx.py" could fix the problem.

IsraelLencina, 2021-02-24:
The yolov4-416.cfg have, actually, "batch=1" the problem is that if you make this `profile.set_shape(...)` with min, max, and opt, equal something you will not have dynamic batch you will must have an input ie. (32, 3, 416, 416).

In 'onnx_to_tensorrt.py' that is solvable in that line referenced, but in yolo_to_onnx.py it's not possible, in line:
```
input_tensor = helper.make_tensor_value_info(
            str(layer_name), # name
            TensorProto.FLOAT, # elem_type
            [self.batch_size, channels, height, width] # input 
)
```
Do you know how to solve it to make the onnx dynamic in yolo_to_onnx?

jkjung-avt, 2021-02-24:
```
input_tensor = helper.make_tensor_value_info(
            str(layer_name), # name
            TensorProto.FLOAT, # elem_type
            [self.batch_size, channels, height, width] # input 
)
```

You'd want to set "batch=-1" in that line of code.  That means the batch dimension is set to "dynamic" in the ONNX model.

In addition, you'd need to use a TensorRT plugin which inherits from "IPluginV2DynamicExt".  You could refer to the plugin implementation in my "batched_yolo" branch: https://github.com/jkjung-avt/tensorrt_demos/tree/batched_yolo/plugins

IsraelLencina, 2021-02-26:
Hi!, it works, with the yololayer_plugin.cu and .h of batched_yolo let me create with dynamics batch size, the only problem is triying to generate the engine of 'yolov4-csp-512.cfg' assert on .cu fileline:

`ASSERT(inputs[0].d[2] == exprBuilder.constant(mYoloHeight));`
Do you know what's happening?

jkjung-avt, 2021-02-26:
No.  I don't think I've tested that cfg with batched_yolo.

closed 2021-02-28 by jkjung-avt

philipp-schmidt, 2021-08-18:
Hello, you can get dynamic batch size up to a max batch size without all these changes. I've already done it similarly in c++.
I'll make a PR.

philipp-schmidt, 2021-08-18:
Well nevermind...
"This version of the ONNX parser only supports TensorRT INetworkDefinitions with an explicit batch dimension. Please ensure the network was created using the EXPLICIT_BATCH NetworkDefinitionCreationFlag."

@jkjung-avt Is there a way around this?

jkjung-avt, 2021-08-19:
> Is there a way around this?

I'm not sure about what you're asking.

To create a TensorRT engine with dynamic batch size, you would set the batch dimension to `-1` in the ONNX model and then set TensorRT optimization profile accordingly (min/max batch sizes for the engine).

philipp-schmidt, 2021-08-19:
If we could use IMPLICIT batching, dynamic batch size would be easier.

But I have implemented a version of what you propose in the merge request. Changing the plugin is probably the biggest change.
And validating for different models.