jkjung-avt/tensorrt_demos #358: KeyError:'size'

lurenlym, 2021-02-26:
I want to convert  my custom yolov4 model int8,
but it appears the keyerror
I test the default yolov4.cfg ,it works.

`
root@aaeon-desktop:/workspace/lym/tensorrt_demos/yolo# python yolo_to_onnx.py -c 11 -m yolov4-custom-608
Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 973, in <module>
    main()
  File "yolo_to_onnx.py", line 956, in main
    verbose=True)
  File "yolo_to_onnx.py", line 433, in build_onnx_graph
    major_node_specs = self._make_onnx_node(layer_name, layer_dict)
  File "yolo_to_onnx.py", line 506, in _make_onnx_node
    node_creators[layer_type](layer_name, layer_dict)
  File "yolo_to_onnx.py", line 564, in _make_conv_node
    kernel_size = layer_dict['size']
KeyError: 'size'
`

my custom yolov4.cfg
`[net]
batch=1
subdivisions=16
width=608
height=608
channels=3
momentum=0.949
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=4000
max_batches = 10000
policy=steps
steps=8000,9000
scales=.1,.1

#cutmix=1
mosaic=1

#:104x104 54:52x52 85:26x26 104:13x13 for 416

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=mish

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=mish

[convolutional]
batch_normalize=1
filters=64
size=1
[... 1074 lines cut ...]
size=1
stride=1
pad=1
filters=48
activation=linear

[yolo]
mask = 6,7,8
anchors = 12, 16, 19, 36, 40, 28, 36, 75, 76, 55, 72, 146, 142, 110, 192, 243, 459, 401
classes=11
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1
scale_x_y = 1.05
iou_thresh=0.213
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
nms_kind=greedynms
beta_nms=0.6
max_delta=5

`

jkjung-avt, 2021-02-26:
I suspect that the yolo cfg parser code failed to handle your custom cfg file because you did not put "#" in front of the comment lines.  For example, these are not valid layer/parameter specs and should be commented with "#".

    Downsample

or

    SPP
    ......
    End SPP

You could do a "diff" with the original [yolov4.cfg](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4.cfg) and find/correct all instances of those.

lurenlym, 2021-02-26:
> I suspect that the yolo cfg parser code failed to handle your custom cfg file because you did not put "#" in front of the comment lines. For example, these are not valid layer/parameter specs and [...]

Thanks, it works now.
The cfg parser code failed to handle the blank line between
`[convolutional]

batch_normalize=1
`

jkjung-avt, 2021-03-01:
I have updated to code to handle empty lines in darknet cfg files more robustly.  On the other hand, I still think you should put in "#" for comments in your cfg files.

closed 2021-03-01 by jkjung-avt

lurenlym, 2021-03-02:
> I have updated to code to handle empty lines in darknet cfg files more robustly. On the other hand, I still think you should put in "#" for comments in your cfg files.

I see！I have put # for comments in my cfg files. Thanks so much!