jkjung-avt/tensorrt_demos #359: TypeError: buffer is too small for requested array

ancoca13, 2021-02-26:
Hello,

I have trained a custom model with 9 classes based on yolov4-416. I have tested it with darknet and it works, but when I try to pass it to onix I get the following error:

$ python3 yolo_to_onnx.py -m yolov4-416 -c 9
Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 973, in <module>
    main()
  File "yolo_to_onnx.py", line 956, in main
    verbose=True)
  File "yolo_to_onnx.py", line 447, in build_onnx_graph
    weight_loader = WeightLoader(weights_file_path)
  File "yolo_to_onnx.py", line 278, in __init__
    self.weights_file = self._open_weights_file(weights_file_path)
  File "yolo_to_onnx.py", line 343, in _open_weights_file
    buffer=weights_file.read(length_header * 4))
TypeError: buffer is too small for requested array

I have attached my cfg file for more information.

can any one help?

Thanks!

[yolov4-416.cfg.txt](https://github.com/jkjung-avt/tensorrt_demos/files/6049013/yolov4-416.cfg.txt)

jkjung-avt, 2021-02-28:
This usually happens when you don't mark the correct layers as outputs in your custom model.  Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/334#issuecomment-765809470.

ancoca13, 2021-02-28:
Hi @jkjung-avt,

Thanks for answering.
I saw that changes needed on the code in other issues, but my cfg file has not any change on the layers. The only thing that I have changed  is the classes number and the number of filters in the convolutional layers before the yolo layers to adapt to my number of classes and y thought that 'c = (args.category_num + 5) * 3' covered yet this change. Am I wrong? What exactly do I have to check to do the changes? I do not really get the logic in them.

Regards,
Antonio.

jkjung-avt, 2021-03-01:
> The only thing that I have changed is the classes number in the yolo layers and the number of filters in the convolutional layers before them

This is correct.

I've just checked your cfg file.  I didn't see anything wrong with it.  I don't have other suggestions than making sure you have named the weights exactly the same as the cfg file: "yolov4-416.weights" and "yolov4-416.cfg".

ancoca13, 2021-03-01:
Hello, 

The name of the weights file was the same and I have added some traces to the code, and it was entering the correct places.  However, I realized that the problem appeared when setting the weight file to an array, so the problem surely was in the weights file. I uploaded again the weights file to the Jetson and it worked! So finally the problem was that the weights file was corrupted.

Thanks for your help!

closed 2021-03-01 by ancoca13