jkjung-avt/tensorrt_demos #360: terminate called after throwing an instance of 'pwgen::PwgenException'   what():  Driver error:

Darshcg, 2021-02-28:
Hi @jkjung-avt,

During conversion from onnx to TRT I am getting an error as: **terminate called after throwing an instance of 'pwgen::PwgenException'   what():  Driver error**. May I know the cause of this error? 

Thanks,
Darshan

jkjung-avt, 2021-02-28:
I have not seen this error before.  I googled the error message and found the following references, based on which I think you should check your CUDA driver/libraries installation.

1. https://forums.developer.nvidia.com/t/pwgenexception-when-building-a-cuda-engine/154469/6
2. https://github.com/stereolabs/zed-docker/issues/28

Darshcg, 2021-02-28:
Hi @jkjung-avt ,

Thank you very much for your reply.
I reinstalled cuda driver, cuda, cudnn, and tensorrt and it worked perfectly.

Thanks,
Darshan

closed 2021-02-28 by Darshcg