jkjung-avt/tensorrt_demos #361: Problem when running on Jetson AGX Xavier

AntonioNuAc, 2021-03-01:
Hello,

When running trt_yolov3.py on Jetson Nano everything goes fine, then, I changed to AGX Xavier and it gives the following errors. Can you please clarify what am I missing?

> OPENBLAS_CORETYPE=ARMV8 python3 trt_yolov3.py --model yolov3-608 --usb --vid 0 --category_num 81

[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[TensorRT] ERROR: coreReadArchive.cpp (38) - Serialization Error in verifyHeader: 0 (Version tag does not match)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
  File "trt_yolov3-custom.py", line 102, in <module>
    main()
  File "trt_yolov3-custom.py", line 88, in main
    trt_yolov3 = TrtYOLOv3(args.model, (yolo_dim, yolo_dim), args.category_num)
  File "/home/omron/Documents/nunant/tensorrt_demos/utils/yolov3.py", line 438, in __init__
    self.context = self._create_context()
  File "/home/omron/Documents/nunant/tensorrt_demos/utils/yolov3.py", line 390, in _create_context
    return self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'
Exception ignored in: <bound method TrtYOLOv3.__del__ of <utils.yolov3.TrtYOLOv3 object at 0x7f614888d0>>
Traceback (most recent call last):
  File "/home/omron/Documents/nunant/tensorrt_demos/utils/yolov3.py", line 446, in __del__
    del self.stream
AttributeError: stream

Thank you very much.

Best regards, 

Antonio.

jkjung-avt, 2021-03-02:
Please refer to the following links.

* https://github.com/NVIDIA/TensorRT/issues/292#issuecomment-580869625
* https://forums.developer.nvidia.com/t/tensorrt-error-rtsafe-corereadarchive-cpp-31-serialization-error-in-verifyheader-0-magic-tag-does-not-match/81872/6

AntonioNuAc, 2021-03-03:
Thanks for the fast reply.
I have to re-generate .trt engine with Xavier, to avoid the TensorRT version problem.

However, I found another issue which is strange since I used your protobuf installer like for Jetson Nano.

> OPENBLAS_CORETYPE=ARMV8 python3 yolov3_to_onnx.py --model yolov3-608

Traceback (most recent call last):
  File "yolov3_to_onnx.py", line 59, in <module>
    import onnx
  File "/usr/local/lib/python3.6/dist-packages/onnx/__init__.py", line 8, in <module>
    from onnx.external_data_helper import load_external_data_for_model, write_external_data_tensors
  File "/usr/local/lib/python3.6/dist-packages/onnx/external_data_helper.py", line 10, in <module>
    from .onnx_pb import TensorProto, ModelProto
  File "/usr/local/lib/python3.6/dist-packages/onnx/onnx_pb.py", line 8, in <module>
    from .onnx_pb2 import *  # noqa
  File "/usr/local/lib/python3.6/dist-packages/onnx/onnx_pb2.py", line 7, in <module>
    from google.protobuf.internal import enum_type_wrapper
ModuleNotFoundError: No module named 'google.protobuf'

I wanted to ask you before trying other things and creating a wrong installation status of my environment.

jkjung-avt, 2021-03-04:
Refer to this: https://github.com/jkjung-avt/tensorrt_demos/issues/130#issuecomment-640221590

You could do `pip3 show protobuf` to see whether the protobuf package has been installed.  Then remove the "\_\_init\_\_.py" in the "google" folder.

In practice, I'd just rename the file.  For example,

```
$ cd /usr/local/lib/python3.6/dist-packages/protobuf-3.8.0-py3.6-linux-aarch64.egg/google
$ sudo mv __init__.py __init__.py.bak
```

closed 2021-03-10 by jkjung-avt