jkjung-avt/tensorrt_demos #363: issues while running yolov4 custom model.

basitraza1214, 2021-03-03:
@jkjung-avt I am having an issue running the custom model. Please find the error output below:
python3 trt_yolo.py --usb 0 --model yolov4-custom-416 --category_num 1
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
Traceback (most recent call last):
  File "/home/basit/project/tensorrt_demos/utils/yolo_with_plugins.py", line 289, in __init__
    allocate_buffers(self.engine)
  File "/home/basit/project/tensorrt_demos/utils/yolo_with_plugins.py", line 201, in allocate_buffers
    assert size % 7 == 0
AssertionError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "trt_yolo.py", line 119, in <module>
    main()
  File "trt_yolo.py", line 106, in main
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num, args.letter_box)
  File "/home/basit/project/tensorrt_demos/utils/yolo_with_plugins.py", line 291, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources
^CException ignored in: <module 'threading' from '/usr/lib/python3.6/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo_with_plugins.TrtYOLO object at 0x7f90bc7320>>
Traceback (most recent call last):
  File "/home/basit/project/tensorrt_demos/utils/yolo_with_plugins.py", line 298, in __del__

I have also attached my config file for your reference.

[yolov4-custom-416.zip](https://github.com/jkjung-avt/tensorrt_demos/files/6075274/yolov4-custom-416.zip)

jkjung-avt, 2021-03-03:
```
assert size % 7 == 0
AssertionError
```

This error (shape of the yolo model output does not match what's expected) is most likely because you did not mark the correct layers (names) as outputs.  Please refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/334#issuecomment-765809470.

basitraza1214, 2021-03-03:
thank  you so much, let me check and then I will get back to you

basitraza1214, 2021-03-03:
Dear @jkjung-avt I am not using tiny yolo, still are the modifications done in referenced comment required. The changes suggested are already made except those for tiny yolo (do I need to make those) but the error persists.
Secondly, I am getting an error when I run onnx_to_tensorrt.py. the error is following
basit@basit-desktop:~/project/tensorrt_demos/yolo$ python3 onnx_to_tensorrt.py --m yolov4-custom-416 --category_num 1
usage: onnx_to_tensorrt.py [-h] [-v] --model MODEL
onnx_to_tensorrt.py: error: unrecognized arguments: --category_num 1

What I did here as a workaround was to remove the --category_num 1 from the command and it worked fine. I do hope that this is not the cause of the assertion error.

jkjung-avt, 2021-03-04:
```
basit@basit-desktop:~/project/tensorrt_demos/yolo$ python3 onnx_to_tensorrt.py --m yolov4-custom-416 --category_num 1
usage: onnx_to_tensorrt.py [-h] [-v] --model MODEL
onnx_to_tensorrt.py: error: unrecognized arguments: --category_num 1
```

This **should not happen**.  Here's the relevant source code in "onnx_to_tensorrt.py":

https://github.com/jkjung-avt/tensorrt_demos/blob/94d06b356822dd230a969d8697dc8c3e55f0dfd6/yolo/onnx_to_tensorrt.py#L167-L169

Please make sure you pull the latest code in this repo.  You need to specify "--category_num 1" (or just "-c 1") for all of "yolo_to_onnx.py", "onnx_to_tensorrt.py" and "trt_yolo.py".

basitraza1214, 2021-03-04:
@jkjung-avt Thank you very much for your response, I will git pull the latest code and give a try

basitraza1214, 2021-03-06:
Thank you very much @jkjung-avt, turns out git pulling the latest code worked and everything is working perfectly. Thank you so much once again.

closed 2021-03-06 by basitraza1214