jkjung-avt/tensorrt_demos #366: Passing multiple frames to engine

IsraelLencina, 2021-03-04:
Hi, i'm triying to pass multiple frames to the engine "TrtYOLO" is it possible?

jkjung-avt, 2021-03-04:
Yes.  Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/306 and https://github.com/jkjung-avt/tensorrt_demos/issues/354.

You will have to modify the source code by yourself.

IsraelLencina, 2021-03-05:
Thank you, i've finally followed the both issues and changed TRTYolo with you directions in #306 

> 1.    Make sure you set max_batch_size when you build the TensorRT engine.
> 2.    allocate_buffers() would then allocate the correct sizes of input/output GPU buffers for you.
> 3.    In TrtYOLO.detect(), stack a batch of preprocessed images into 1 contiguous numpy array, shaped as (batch_size, 3, input_height, input_width).
> 4.    do_inference_v2() would do the right thing for you. No modification of code is needed here.
> 5.    Post-process outputs of the TensorRT engine, one image at a time. Note that the 2 or 3 outputs would also contain batched outputs for all input images.

The problematic step have been 4, the cuda size and dtype was problematic, finally i've solved watching that "np.ascontiguousarray" convert the data to float64 but the input expects float32, then the memory overflow, i leave down the link in nvidia to figure out this issue.

https://forums.developer.nvidia.com/t/pycuda-driver-logicerror-cumemcpyhtodasync-failed-invalid-argument/83132

Don't close still, i'm finishing the implementation.

IsraelLencina, 2021-03-05:
As i've seen in fifth point:

> Post-process outputs of the TensorRT engine, one image at a time. Note that the 2 or 3 
> outputs would also contain batched outputs for all input images.

I've changed the dims of inputs and outputs the result in case of batch 2 is correct (ie: yolov4-416 -> 1038336 for input (2, 3, 416, 416) and outputs are (56784*2, 14196
2, 3549*2)) so it should be correct, had sense.
But when i call trtYolo.detect() return trtoutput with the result of this command:
`outputs[0].host[int(113568/2):len(outputs[0].host)]` (result for the second image batched) all to 0.

PD: I'm using the code of 'batched_yolo' branch.

jkjung-avt, 2021-03-05:
> The problematic step have been 4, the cuda size and dtype was problematic, finally i've solved watching that "np.ascontiguousarray" convert the data to float64 but the input expects float32, then the memory overflow

This should not be the case.  The "dtype" of the numpy arrays should all be "float32", and "np.ascontiguousarray()" and "cuda.memcpy_htod()" should work without any problem.  You could see very similar code in my INT8 calibrator implementation:

https://github.com/jkjung-avt/tensorrt_demos/blob/94d06b356822dd230a969d8697dc8c3e55f0dfd6/yolo/calibrator.py#L140

jkjung-avt, 2021-03-05:
> I'm using the code of 'batched_yolo' branch.

My "batched_yolo" branch is experimental.  I'm not maintaining code in that branch.  If you don't really need a TensorRT engine with dynamic batches, I suggest you to stick with the "master" branch.

I have verified that my yolo_layer plugin could handle batched inputs correctly, so does the TensorRT yolov4 engine.

Otherwise, as stated earlier, you will have to work out the inference code (for batched image inputs) by yourself.

IsraelLencina, 2021-03-08:
The problem was in TRTYolo line: 
`assert self.context.set_binding_shape(0, input_shape)`
I was putting input_shape[0] = 1 but it should be max_batch_size, now the engine had totally dynamic shape. Thanks for all, i will merge to develop soon.

closed 2021-03-08 by IsraelLencina

liermvl, 2021-03-08:
@IsraelLencina 

Do you have an example of how you handle step 5: Post-process outputs of the TensorRT engine, one image at a time. ?
Would be really nice if you can share some code.