jkjung-avt/tensorrt_demos #367: Demo 5: onxx_to_tensorrt.py failed with custom YOLOv4 model

DoVan-Thuan, 2021-03-05:
Hi, @jkjung-avt. Thank you very much for your repo.

I'm trying to convert my custom yolov4 model to yolov4-288 and i fail with error:

Building an engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
python3: yolo_layer.cu:119: virtual nvinfer1::Dims nvinfer1::YoloLayerPlugin::getOutputDimensions(int, const nvinfer1::Dims*, int): Assertion `inputs[0].d[1] == mYoloHeight' failed.
Aborted (core dumped)

1. I only changed [filters=255] to [filters=27] in the 3 [convolutional] before each [yolo] layer in the config file in training because i have 4 classes. Here is my config file: 
[yolov4-vcc.cfg.txt](https://github.com/jkjung-avt/tensorrt_demos/files/6091292/yolov4-vcc.cfg.txt)
2. Then, i created the yolov4-vcc-288.cfg and yolov4-vcc-288.weights with the following commands:
`cat yolov4-vcc.cfg | sed -e '2s/batch=64/batch=1/' | sed -e '7s/width=608/width=288/' | sed -e '8s/height=608/height=288/' > yolov4-vcc-288.cfg`
and
`ln -sf yolov4-vcc.weights yolov4-vcc-288.weights`

I run the yolo_to_onxx.py successfully but failed in running onxx_to_tensorrt.py. Could you help me solve this problem?
Thank you.

jkjung-avt, 2021-03-05:
The "width" and "height" in your cfg file are 416 and 416.  Please replace them with 288 instead.

DoVan-Thuan, 2021-03-06:
> The "width" and "height" in your cfg file are 416 and 416. Please replace them with 288 and run "yolo_to_onnx.py" again.

Thank you so much, problem is solved. By the way, i have some questions:

1. I saw in your README.md, your yolov4-288(FP16) ran on jetson nano got 7.93 but for me it only ~3.5-3.6 FPS. I ran with 1920x1080 video. Does it affect the FPS and are there any way to improve FPS ?
2. I added deepsort for tracking and FPS dropped to ~2.5-2.6 FPS. Can we convert deepsort to TensorRT ? Because i want the fps is 5 at least with detection + tracking.

Thank you very much.

jkjung-avt, 2021-03-06:
> I saw in your README.md, your yolov4-288(FP16) ran on jetson nano got 7.93FPS but for me it only ~3.5-3.6 FPS. I ran with 1920x1080 video. Does it affect the FPS and are there any way to improve FPS ?

My FPS measurement was done with [dog.jpg](https://raw.githubusercontent.com/pjreddie/darknet/master/data/dog.jpg).  You could compare the FPS difference between dog.jpg and your 1920x1080 video.

> I added deepsort for tracking and FPS dropped to ~2.5-2.6 FPS. Can we convert deepsort to TensorRT ? Because i want the fps is 5 at least with detection + tracking.

I have not looked into deepsort so I cannot give any advice.

DoVan-Thuan, 2021-03-06:
> My FPS measurement was done with [dog.jpg](https://raw.githubusercontent.com/pjreddie/darknet/master/data/dog.jpg). You could compare the FPS difference between dog.jpg and your 1920x1080 video.
>
>
> I [...]

Does the size of input image/video affect the fps ? How can i improve the fps ?

jkjung-avt, 2021-03-06:
> Does the size of input image/video affect the fps ?

It would take a little more time for resizing images (1920x1080 -> 288x288) in preprocessing, but the difference should not be too much.

DoVan-Thuan, 2021-03-06:
> > Does the size of input image/video affect the fps ?
> 
> It would take a little more time for resizing images (1920x1080 -> 288x288) in preprocessing, but the difference should not be too much.

Thank you so much for your support, i'm gonna close this issue.

closed 2021-03-06 by DoVan-Thuan

Myron1996, 2021-03-08:
Hi @jkjung-avt @DoVan-Thuan , I trained a custom Yolov4 model and it generated the checkpoints file instead of .WEIGHTS file . Can you please suggest me if we can use those instead of .weights file. Thank you so much for your work . It helps a lot to newcomers like me.