jkjung-avt/tensorrt_demos #369: Custom Checkpoints file from YOLOV4

Myron1996, 2021-03-08:
Hi @jkjung-avt , I trained a `custom Yolov4` model and it generated the **checkpoints** file instead of `.WEIGHTS` file . Can you please suggest me if we can use those instead of `.weights` file. Thank you so much for your work . It helps a lot to newcomers like me.

jkjung-avt, 2021-03-08:
My TensorRT yolov4/yolov3 implementation only supports custom models in darknet format (cfg & weights).  If you want to use code in this repo, you could try to use tools which convert other (PyTorch or TensorFlow) models into darknet format.

Myron1996, 2021-03-10:
HI @jkjung-avt . thanks for your quick response. Can you please elaborate what exactly you mean by **`tools`** ? Thank you

jkjung-avt, 2021-03-10:
For example, @Darshcg has successfully converted the "YOLOv3-SPP-ultralytics" (pytorch) model to onnx, and built the TensorRT engine with code in this repo.  Refer to [this](https://github.com/jkjung-avt/tensorrt_demos/issues/267#issuecomment-718145288) and [this](https://github.com/jkjung-avt/tensorrt_demos/issues/267#issuecomment-721714934).

You might refer to his solution or discuss with him.

Myron1996, 2021-03-10:
Okay I will  look into that . Thanks :+1: I will post the progress in here so that others can benefit form it too.

jkjung-avt, 2021-03-19:
There's another success case of pytorch model converted to TensorRT and inferenced with code in this repo: https://github.com/jkjung-avt/tensorrt_demos/issues/375#issuecomment-797190610

I'm closing this case now.

closed 2021-03-19 by jkjung-avt