jkjung-avt/tensorrt_demos #370: suggestion for gstreamer inputs

Tetsujinfr, 2021-03-08:
hi
just a suggestion: instead of hardcoding the gstreamer command in the utils/camera.py module, I think it could be more flexible to let the user specify the gstreamer string as a command line parameter, to give more flexibility.

For instance, I did a quick and dirty modification in the attached files, adding a "gstr" param flag which has to be followed by the gstreamer string between quotes.
[camera.zip](https://github.com/jkjung-avt/tensorrt_demos/files/6104140/camera.zip)

jkjung-avt, 2021-03-12:
Thanks for the suggestion.  I have added this option into the code: https://github.com/jkjung-avt/tensorrt_demos/commit/bb4b463a756b8d7f88fa1f056efe1ac1a1cf5b5d

closed 2021-03-12 by jkjung-avt

Tetsujinfr, 2021-03-12:
Thanks a lot, really cool.