jkjung-avt/tensorrt_demos #375: pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument

freeluojing123, 2021-03-11:
I trained the YOLOV3 model using the PyTorch model and successfully converted it to ONNX. Using your code, I converted ONNX to TRT, but the following problems occurred when making predictions
[image]

freeluojing123, 2021-03-11:
`nano@nano-desktop:~/work/tensorrt_demos$ python3 trt_yolov3.py --model yolov3-wheat-new-416 --image wheat.jpg
Traceback (most recent call last):
  File "trt_yolov3.py", line 96, in <module>
    main()
  File "trt_yolov3.py", line 88, in main
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)
  File "trt_yolov3.py", line 56, in loop_and_detect
    boxes, confs, clss = trt_yolov3.detect(img, conf_th)
  File "/home/nano/work/tensorrt_demos/utils/yolov3.py", line 468, in detect
    stream=self.stream)
  File "/home/nano/work/tensorrt_demos/utils/yolov3.py", line 379, in do_inference_v2
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
  File "/home/nano/work/tensorrt_demos/utils/yolov3.py", line 379, in <listcomp>
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument`

freeluojing123, 2021-03-11:
This is important to me. Please help me

jkjung-avt, 2021-03-12:
Refer to this: https://forums.developer.nvidia.com/t/pycuda-driver-logicerror-cumemcpyhtodasync-failed-invalid-argument/83132/2

You should make sure your model is indeed taking 416x416 input.  Next, try printing out the sizes of "in.device" and "in.host" buffers and check why they differ.

Otherwise, I never tested my own code with PyTorch yolov4/yolov3 models.  I don't guarantee it would work...

freeluojing123, 2021-03-12:
> Refer to this: https://forums.developer.nvidia.com/t/pycuda-driver-logicerror-cumemcpyhtodasync-failed-invalid-argument/83132/2
>
> You should make sure your model is indeed taking 416x416 input. Next, [...]

I solved this problem. When converting the pth model to the onnx model, the input size was wrong. thank you very much

freeluojing123, 2021-03-12:
The pth model is converted to the onnx model, and then the trt model is converted（Take yolov3 as an example）. The whole process has been completed, and I will post it in a blog.

jkjung-avt, 2021-03-18:
It'd be really great if you could share the experience in a blog post.  I will close this issue now.

closed 2021-03-18 by jkjung-avt