jkjung-avt/tensorrt_demos #377: Mtcnn with multiple faces

VLorin, 2021-03-15:
Hi, first of all great work! I'm working on Jetson nano and your implementation is the best I've seen for frontal face detection.But I have an issue (not sure if you can help), but I'm trying to do a face recognition using your mtcnn as a face detector, but sometime when I launch he can't see more than one or two faces at the time:
[image]
But some other time it works perfectly fine :
[image]
Between the two pictures I've just stopped and restarted the same program without making any modification. 
Can you help me with this ? Thx
I've didn't modify your code related to the mtcnn, I just use `mtcnn.detect()` in my code

jkjung-avt, 2021-03-16:
I think you are facing a known issue as described here: https://github.com/jkjung-avt/tensorrt_demos/issues/313#issuecomment-787648496

Please try to use a different version of TensorRT (don't use 7.1.3.4).

VLorin, 2021-03-16:
Thx you for your response ! Will try soon

closed 2021-03-22 by jkjung-avt