jkjung-avt/tensorrt_demos #379: yolov4 tensorrt with ROS : (pycuda error) context stack was not empty upon module clean

indra4837, 2021-03-18:
Hi, great work on your tensorrt_demos! You can find my code [here](https://github.com/indra4837/yolov4_trt_ros)

I adapted your work to ROS for my project use and achieve great results with it. However, I am always receiving a pycuda error after the program is killed. Do refer to the screenshot for a sample error log found. 

[image]

jkjung-avt, 2021-03-18:
@indra4837  Could you provide a simpler example (say, with only minor modifications to my original trt_yolo.py) to reproduce this problem?

jkjung-avt, 2021-03-30:
Closing due to no response.

closed 2021-03-30 by jkjung-avt