jkjung-avt/tensorrt_demos #380: Problem when installing bazel 0.15.2

DoVan-Thuan, 2021-03-22:
Hi @jkjung-avt, thank you for your repo.

I'm trying to run ssd on jetson nano and i followed your blog to install tensorflow 1.12.2 https://jkjung-avt.github.io/build-tensorflow-1.12.2/

But when i failed to install bazel 0.15.2 with this error:
🍃  Building Bazel from scratch...unzip -qn third_party/asm/asm-6.0-sources.jar -d /tmp/bazel_mnf2xFva/classes
error: invalid zip file with overlapped components (possible zip bomb)
cp: cannot stat 'output/bazel': No such file or directory

Could you help me solve this issue ? Thank you very much !

jkjung-avt, 2021-03-22:
>unzip -qn third_party/asm/asm-6.0-sources.jar -d /tmp/bazel_mnf2xFva/classes
> error: invalid zip file with overlapped components (possible zip bomb)

Please check if the file (third_party/asm/asm-6.0-sources.jar) is corrupted.  Try deleting the ${HOME}/src/bazel* folders/files and running the sccript again.

DoVan-Thuan, 2021-03-23:
> Please check if the file (third_party/asm/asm-6.0-sources.jar) is corrupted. Try deleting the ${HOME}/src/bazel* folders/files and running the sccript again.

Thank for your response. I checked the asm-6.0-sources.jar and i could unzip it but get the error: "invalid zip file with overlapped components (possible zip bomb)". I also deleted the ${HOME}/src/bazel* folders/files and re-run the script but i still got that problem. 
Could you help me figure out what problem is ? Thank you very much.

jkjung-avt, 2021-03-24:
I've never encountered such a problem.  And I don't have a solution for you.  Sorry...

DoVan-Thuan, 2021-03-31:
I didn't follow your block to set up the environment. I installed tensorflow 1.15 and i ran your ssd successfully. However, the fps is very low, it's only ~3.7.
Can you explain it for me ? Thank you very much !

jkjung-avt, 2021-03-31:
@DoVan-Thuan  If you start with a clean JetPack environment and set up your Jetson system the same way as me (e.g. [JetPack-4.4 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.4/), then you should get similar FPS numbers as I've posted.

Otherwise, I'm not able to guess why you get a different result as mine.  It could be due to a lot of different reasons.  For example, difference in system libraries, pip packages, hardware devices/configurations, or even due to some other memory-hugging applications.

I don't know what else I need to explain...

closed 2021-04-10 by jkjung-avt