jkjung-avt/tensorrt_demos #381: mAP is much lower in  YOLOv3 INT8 model

SAIL-LAB155, 2021-03-24:
**Hi, this is the eval result of YOLOv3 TRT model with FP16**
[image]
**This is  the eval result of YOLOv3 TRT model with INT8**
[image]
**And my dataset is about 4k images, and I use  the val dataset for calibration which is about 400 images.
Do you know why mAP drops sharply? Thx.**

jkjung-avt, 2021-03-24:
No, I don't have any idea about what you've described.  Maybe you could look at a few example images (from your val set) to see why the INT8 engine performs poorly.

Sean-hku, 2021-03-25:
Do u mean I should use part of val set rather than whole val set?

jkjung-avt, 2021-03-25:
No, I suggest running "trt_yolo.py" and visualize the resulting detections (bounding boxes).  See whether the bad mAP is due to missing detections or false positives or else, and try to figure out the next debugging step from there.

Sean-hku, 2021-03-25:
Thank you for your quick reply. I will try it.

SAIL-LAB155, 2021-03-26:
I find there are lots of missing detections in int8 models. What can I do to improve the mAP?
Thx

jkjung-avt, 2021-03-26:
Honestly, this would be difficult to debug.  You could try:

1. set a lower "conf_th" (confidence threshold) to see if you get better detection results,

    https://github.com/jkjung-avt/tensorrt_demos/blob/d6ae0d7cc7fc356054c896eb5f9b0873d5d5ab6a/trt_yolo.py#L101

2. check if NVIDIA's new [Polygraphy](https://github.com/NVIDIA/TensorRT/tree/master/tools/Polygraphy) tool could be used to debug the TensorRT INT8 model.  You could refer to https://github.com/jkjung-avt/tensorrt_demos/issues/315#issuecomment-805419258 for more information.

jkjung-avt, 2021-03-28:
I would also suggest you to try use as many images as possible (e.g. use your whole training set and test set) for INT8 calibration.  If that helps to improve mAP of the INT8 engine, then you know the problem lies in calibration dataset.

closed 2021-04-05 by jkjung-avt