jkjung-avt/tensorrt_demos #382: Cannot convert retrained ssd_inception_v2 model

avanish, 2021-03-24:
I have retrained `ssd_inception_v2_coco` model with custom dataset, and exported the model using `export_inference_graph.py` in Tensorflow Object Detection API.

However, when I try to convert the model to TensorRT using the code provided here, I get this error.

```
2021-03-24 11:33:25.560976: I tensorflow/stream_executor/platform/default/dso_loader.cc:48] Successfully opened dynamic library libcudart.so.10.2
WARNING:tensorflow:Deprecation warnings have been disabled. Set TF_ENABLE_DEPRECATION_WARNINGS=1 to re-enable them.
NOTE: UFF has been tested with TensorFlow 1.15.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.9
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 300
      }
      dim {
        size: 300
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:226: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: GridAnchor_TRT yet.
Converting MultipleGridAnchorGenerator as custom op: GridAnchor_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:143] Marking ['NMS'] as outputs
No. nodes: 811
UFF Output written to /home/nano/workspace/tensorrt_demos/ssd/ssd_inception_v2_coco.uff
UFF Text Output written to /home/nano/workspace/tensorrt_demos/ssd/ssd_inception_v2_coco.pbtxt
[TensorRT] INFO: Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output.
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cpp,249
Aborted (core dumped)
```

I have noticed we have higher no. of nodes (811) compared to what we get in base model (5XX).

Thanking in advance.

jkjung-avt, 2021-03-24:
Which version of tensorflow are you using?  (The code would only work for tensorflow 1.x.)

avanish, 2021-03-24:
Tensorflow is 1.x. Figured out the issue.

Turns out `numClasses` was +1 for background. So for 5 classes, I had to use `numClasses: 6`.

closed 2021-03-24 by avanish