jkjung-avt/tensorrt_demos #385: int8 problem

QZ-cmd, 2021-03-29:
Hello, I converted int8-yolov3 after testing map = 0,It's normal for me to turn fp16， the original onnx output has two：output_loc and output_conf, now after calibration int8 two more output, I suspect is the problem of model, do you know how to solve it?Looking forward to your reply, thank you！
#######################################################
calib_yolov3-int8darknetbaseline-416.bin
output_loc: 3cc41874
(Unnamed Layer* 315) [Shuffle]_output: 3d0743b7
(Unnamed Layer* 316) [Softmax]_output: 3c1047bf
output_conf: 3c1047bf

jkjung-avt, 2021-03-29:
Please make sure you have prepared the calibration dataset properly.  In addition, did you see any errors/warning when building the INT8 engine?

QZ-cmd, 2021-03-30:
Thank you for your timely reply！I turned int8 without reporting an error, use  the entire test set calibration or use1000 images from the test set。I saw you answer that the concat node would have an impact, right? I didn't have these two outputs in my original onnx model。I suspect there is a problem here。If you need onnx model please let me know。Looking forward to your answer！
(Unnamed Layer* 315) [Shuffle]_output: 3d0743b7
(Unnamed Layer* 316) [Softmax]_output: 3c1047bf
############################################
TRT-7100-EntropyCalibration2
actual_input_1: 3c001010
445: 3c12acf8
446: 3bc82d2d
447: 3b15afad
.....................
727: 3cc41874
752: 3d0743b7
728: 3cc41874
753: 3d0743b7
765: 3d0743b7
output_loc: 3cc41874
(Unnamed Layer* 315) [Shuffle]_output: 3d0743b7
(Unnamed Layer* 316) [Softmax]_output: 3c1047bf
output_conf: 3c1047bf

jkjung-avt, 2021-03-30:
If you are not building the TensorRT engine from a DarkNet model (cfg/weights), I cannot guarantee my code would work...

For example, my code ([link](https://github.com/jkjung-avt/tensorrt_demos/blob/d6ae0d7cc7fc356054c896eb5f9b0873d5d5ab6a/yolo/plugins.py#L66)) would peek into the cfg file to find anchor values for the yolo_layer plugins.

QZ-cmd, 2021-03-30:
The question is, isn't that part of int8 quantifying that part of it all done by NVIDIA，It should all be the same, not completely open source, I use your code, fp16 is correct, it's really distressing, I don't know where to solve the problem.

jkjung-avt, 2021-03-30:
>  isn't that part of int8 quantifying that part of it all done by NVIDIA

Yes.  My [int8 calibrator](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolo/calibrator.py) is implemented by following NVIDIA documentation and sample code.

I don't know the cause of your problem.  I think you should seek help from NVIDIA.

QZ-cmd, 2021-03-30:
ok，thanks for your time.

closed 2021-04-08 by jkjung-avt