jkjung-avt/tensorrt_demos #386: onnx_to_tensorrt.py fail

Sheepy0, 2021-03-29:
For Demo #5 Step #5
$ python3 onnx_to_tensorrt.py -m yolov4-416

I get the following error:

Loading the ONNX file...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 702 (the launch timed out and was terminated)
[TensorRT] ERROR: ../rtSafe/safeRuntime.cpp (32) - Cuda Error in free: 702 (the launch timed out and was terminated)
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted (core dumped)

I  suspected that it was a swap memory issue, and increased it. 
"free -m" now shows that swap has 8000 free.

error remains.

Thank you !

jkjung-avt, 2021-03-30:
Here's what [NVIDIA documentation](https://docs.nvidia.com/cuda/cuda-runtime-api/group__CUDART__TYPES.html#group__CUDART__TYPES_1g3f51e3575c2178246db0a94a430e0038) says about CUDA Error 702:

> cudaErrorLaunchTimeout = 702
> This indicates that the device kernel took too long to execute. This can only occur if timeouts are enabled - see the device property kernelExecTimeoutEnabled for more information. This leaves the process in an inconsistent state and any further CUDA work will return the same error. To continue using CUDA, the process must be terminated and relaunched.

And this error is usually associated with HW or driver issues.  I'd suggest you to double check your CUDA driver/toolkit, cuDNN and TensorRT installation.  Otherwise, referring to [this](https://render.otoy.com/forum/viewtopic.php?f=30&t=63360), you might try to disable X (offloading GPU from graphics rendering tasks) and then do onnx_to_tensorrt.py in text mode.

To enter text mode (on a Ubuntu system), use the following command:

```
$ sudo systemctl isolate multi-user.target
```

And to get back to graphical mode:

```
$ sudo systemctl isolate graphical.target
```

Sheepy0, 2021-04-05:
for some reason, using "-v" to enable verbose logging fixed the issue.

many thanks.

closed 2021-04-05 by Sheepy0

STAROFWIND, 2021-05-13:
I am also stuck in this issue. I am not lucky when doing -v like  Mr.Sheepy0 . So can anyone suggest for solution of above problem ??? Thank you so much !!!

Myron1996, 2021-06-02:
I also have the same issue 
```
tdc@tdc:~/Downloads/tensorrt_demos/yolo$ python3 onnx_to_tensorrt.py -m yolov4-tiny-416
Loading the ONNX file...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 702 (the launch timed out and was terminated)
[TensorRT] ERROR: ../rtSafe/safeRuntime.cpp (32) - Cuda Error in free: 702 (the launch timed out and was terminated)
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted (core dumped)
```
Any Success ? @STAROFWIND @jkjung-avt