jkjung-avt/tensorrt_demos #387: Diference with original yolo

Whitefox99, 2021-04-04:
Hello sir, is the detection show any different result in model in tensorrt version compare to original yolo model? is it less accurate? or more accurate? or any diferece detection result? thank you

jkjung-avt, 2021-04-05:
Detection results of the TensorRT engines might differ from the original DarkNet YOLO models.  One possible source of this discrepancy could be difference in FP16 (TensorRT) and FP32 (DarkNet) calculations.  I also suspect there are still issues in my DarkNet -> ONNX -> TensorRT conversion (refer to this discussion: https://github.com/jkjung-avt/tensorrt_demos/issues/315).

Otherwise, I have provided mAP evaluation results of the TensorRT engines in README.md.  Please read it.

Whitefox99, 2021-04-05:
is there a way that i can compare original yolo and tensorrt version head to head and get loss function after the conversion?

jkjung-avt, 2021-04-05:
I only implemented mAP evaluation (using "pycocotools") for the TensorRT engines.  You need to modify the code by yourself if you want to compare detection results between the DarkNet model and the corresponding TensorRT engine.

Whitefox99, 2021-04-05:
do you have guide for me to modify the code?

jkjung-avt, 2021-04-05:
No.  Please work on that by yourself.

Whitefox99, 2021-04-05:
Ok, Thank you sir

closed 2021-04-05 by Whitefox99