jkjung-avt/tensorrt_demos #388: onnx configuration and int8 calib

maidmentdaniel, 2021-04-13:
Hi,

I'm attempting to build a multi-batch Tensorrt engine from an onnx file, in this case a custom trained yolov4 with only two classes. I've verified that I can build fp16 engines with dynamic batch sizing, but can't quite get the calibration of int8 right. Do you have any advice? Specifically, should the onnx file be dynamic? And, if so,  what is the process for calibration? If not, should it be static of batch size > 1 or batch_size = 1?

Further, do you have any experience with Tensorrt's Polygraphy tool?

Best,

Daniel

jkjung-avt, 2021-04-14:
> I've verified that I can build fp16 engines with dynamic batch sizing, but can't quite get the calibration of int8 right. Do you have any advice? Specifically, should the onnx file be dynamic? And, [...]

I myself have only tried to build FP16 engines with dynamic batch.  I would set the batch dimension to "-1" (dynamic) in the ONNX model.  Then I'd create a TensorRT optimization profile with the proper batch range.  For example, to build a TensorRT engine with dynamic batch size ranging from 1 to 4:

```
            # MAX_BATCH_SIZE = 4
            profile.set_shape(
                '000_net',                          # input tensor name
                (1, 3, net_h, net_w),  # min shape
                (MAX_BATCH_SIZE, 3, net_h, net_w),  # opt shape
                (MAX_BATCH_SIZE, 3, net_h, net_w))  # max shape
```

I haven't tried it for INT8 engines.  Please share your experience once you get the calibrator to work.

> Further, do you have any experience with Tensorrt's Polygraphy tool?

NVIDIA Polygraphy is easy to use (although documentation is lacking).  You could refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/315#issuecomment-817249830

maidmentdaniel, 2021-04-14:
> > I've verified that I can build fp16 engines with dynamic batch sizing, but can't quite get the calibration of int8 right. Do you have any advice? Specifically, should the onnx file be dynamic? And, if so, what is the process for calibration? If not, should it be static of batch_size > 1 or batch_size = 1?
> 
> I myself have only tried to build FP16 engines with dynamic batch. I would set the batch dimension to "-1" (dynamic) in the ONNX model. Then I'd create a TensorRT optimization profile with the proper batch range. For example, to build a TensorRT engine with dynamic batch size ranging from 1 to 4:
> 
> ```
>             # MAX_BATCH_SIZE = 4
>             profile.set_shape(
>                 '000_net',                          # input tensor name
>                 (1, 3, net_h, net_w),  # min shape
>                 (MAX_BATCH_SIZE, 3, net_h, net_w),  # opt shape
>                 (MAX_BATCH_SIZE, 3, net_h, net_w))  # max shape
> ```
> 
> I haven't tried it for INT8 engines. Please share your experience once you get the calibrator to work.
> 
> > Further, do you have any experience with Tensorrt's Polygraphy tool?
> 
> NVIDIA Polygraphy is easy to use (although documentation is lacking). You could refer to: [#315 (comment)](https://github.com/jkjung-avt/tensorrt_demos/issues/315#issuecomment-817249830)

Thanks for getting back to me. This is exactly what I tried and thus far nothing, it's getting quite frustrating. Calibration for batch_size 1 works just fine, but anything larger than 1 and I got it working once and then didn't manage to replicate the result. On the occasion that I got it working, it was by generating a calibration file for a batch size of 1, and then using that calibration file to generate the engine using a static onnx batch size of N > 1, naturally it wasn't dynamic.

Thanks for the pointer with Polygraphy.

maidmentdaniel, 2021-04-14:
> I haven't tried it for INT8 engines. Please share your experience once you get the calibrator to work.

I will do so.

jkjung-avt, 2021-05-31:
Closing this issue for now so that I don't spend time tracking it.

closed 2021-05-31 by jkjung-avt