jkjung-avt/tensorrt_demos #390: configure.py error: no such option: --no-use-shipped-boost

Duarte-Nunes, 2021-04-13:
Im getting this error while trying to install pycuda, any help would be apreciated.

Duarte-Nunes, 2021-04-13:
Solved by deleting siteconf.py

closed 2021-04-13 by Duarte-Nunes

Sam-Fisher-20, 2021-06-28:
Hi - how shall i find siteconf.py as well - facing same error. But unsure where the file is.

jkjung-avt, 2021-06-28:
https://github.com/jkjung-avt/tensorrt_demos/issues/214#issuecomment-686193744

Sam-Fisher-20, 2021-06-28:
Hi - started installing pycuda script in HOME/Somename/ tensorrt_demos/ssd--- now i am looking for src in Home -- file not found . Same thing in Home/Somename. 

When i do whereis siteconf.py - nothing turns up. 
I am using ubuntu 20.04

The solution seems simple enough - if only i could find the file :(

jkjung-avt, 2021-06-28:
The "install_pycuda.sh" script would create the "${HOME}/src/pycuda-2019.1.2" directory for you: https://github.com/jkjung-avt/tensorrt_demos/blob/549d50b167f57f121e2fc8eed214a30d41f19de3/ssd/install_pycuda.sh#L13-L14

You could try to remove the whole directory and run the script again:

```shell
$ rm -rf ${HOME}/src/pycuda-2019.1.2
$ ./install_pycuda.sh
```

Sam-Fisher-20, 2021-06-28:
Thank you @jkjung-avt - found the tree - removed it all and ran the script again.

The next error is unable to locate : -lboost_python*-py3 

I installed python3.8 deb file again to resolve this - but when i deleted the tree and started script again the error came back. Should i install deb file again?

Sam-Fisher-20, 2021-06-28:
This error is solved - so I am happy to close the configure.py error. Will open lboost issue in another thread. Thank you.

pureloveljc, 2021-11-11:
How did you solve it