jkjung-avt/tensorrt_demos #393: Cannot convert onnx to trt (yolo-csp-256 with single class cfg)

ksy961224, 2021-04-19:
Hi Sir,
I have issue with converting modified yolov4-csp-256.onnx to TensorRT format, which has different custom class number (classes=1, filter=18, else not edited at all)
I have no problem with default yolov4-csp-256.cfg to trt file
[yolov4-csp-256-numplate.zip](https://github.com/jkjung-avt/tensorrt_demos/files/6334262/yolov4-csp-256-numplate.zip)

but can't convert modified-yolov4-csp-256.cfg to trt file
During parsing onnx file, Segmentation fault error occurs
I've seen that can be caused by oom, So I did same job on Google Colab.
 But same happens (default onnx can be converted on colab too)

I trained with cfg with 512size, and 
used 256cfg with 512weight file during converting, as same as default code does.

Is there any possible way to solve it?
Anyway, really appreciate for ur code

jkjung-avt, 2021-04-20:
Please rename your DarkNet model files as "yolov4-csp-numplate-256.cfg" and "yolov4-csp-numplate-256.weights".  Then re-run "yolo_to_onnx.py"

```shell
$ python3 yolo_to_onnx.py -c 1 -m yolov4-csp-numplate-256
```

The relevant source code is here: https://github.com/jkjung-avt/tensorrt_demos/blob/4b26770dca33cbe613f65e9b64ec3b9c912e2084/yolo/yolo_to_onnx.py#L946-L949

ksy961224, 2021-04-20:
Thanks a lot. :)
I havent tried converting cfg to onnx with name chaning

closed 2021-04-20 by ksy961224