jkjung-avt/tensorrt_demos #398: Error while build onnx file

gioriog, 2021-04-26:
Hello, I am trying to convert a yolo custom model to an onnx format and use it for a unity application. I run into this error about some Upsample layer. Is there something I can fix from cfg file ?

Checking ONNX model...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 987, in <module>
    main()
  File "yolo_to_onnx.py", line 977, in main
    onnx.checker.check_model(yolo_model_def)
  File "/home/cim40/anaconda3/lib/python3.8/site-packages/onnx/checker.py", line 104, in check_model
    C.check_model(protobuf_string)
onnx.onnx_cpp2py_export.checker.ValidationError: Op registered for Upsample is deprecated in domain_version of 14

==> Context: Bad node spec for node. Name: 086_upsample OpType: Upsample

[yolov3-custom-416.zip](https://github.com/jkjung-avt/tensorrt_demos/files/6376928/yolov3-custom-416.zip)

jkjung-avt, 2021-04-26:
Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/70.  You have to use onnx==1.4.1 for yolo models in thos repo.

> subprocess.CalledProcessError: Command '['/usr/local/bin/cmake', '--build', '.', '--', '-j', '28']' returned non-zero exit status 2.
> 
> ERROR: Command errored out with exit status 1: /home/cim40/anaconda3/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bnjy79yo/onnx_f7afeb7a244042a196acbe119bf210d9/setup.py'"'"'; file='"'"'/tmp/pip-install-bnjy79yo/onnx_f7afeb7a244042a196acbe119bf210d9/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(file) if os.path.exists(file) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' install --record /tmp/pip-record-k9f55x9b/install-record.txt --single-version-externally-managed --compile --install-headers /home/c/anaconda3/include/python3.8/onnx Check the logs for full command output.

What's the exact error you encountered when installing onnx==1.4.1?  Could you dig out the log file?

gioriog, 2021-04-26:
Looking in indexes: https://pypi.org/simple, https://pypi.ngc.nvidia.com
Collecting onnx==1.4.1
  Downloading onnx-1.4.1.tar.gz (2.9 MB)
     |████████████████████████████████| 2.9 MB 5.6 MB/s 
Requirement already satisfied: protobuf in ./anaconda3/lib/python3.8/site-packages (from onnx==1.4.1) (3.15.8)
Requirement already satisfied: numpy in ./anaconda3/lib/python3.8/site-packages (from onnx==1.4.1) (1.19.2)
Requirement already satisfied: six in ./anaconda3/lib/python3.8/site-packages (from onnx==1.4.1) (1.15.0)
Requirement already satisfied: typing>=3.6.4 in ./anaconda3/lib/python3.8/site-packages (from onnx==1.4.1) (3.7.4.3)
Requirement already satisfied: typing-extensions>=3.6.2.1 in ./anaconda3/lib/python3.8/site-packages (from onnx==1.4.1) (3.7.4.3)
Building wheels for collected packages: onnx
  Building wheel for onnx (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/cim40/anaconda3/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-v1vncfxm/onnx_4b4b637c32964d11afe6a11a8d86a354/setup.py'"'"'; __file__='"'"'/tmp/pip-install-v1vncfxm/onnx_4b4b637c32964d11afe6a11a8d86a354/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-t2d3_42g
       cwd: /tmp/pip-install-v1vncfxm/onnx_4b4b637c32964d11afe6a11a8d86a354/
  Complete output (127 lines):
  fatal: not a git repository (or any of the parent directories): .git
  /home/cim40/anaconda3/lib/python3.8/site-packages/setuptools/dist.py:642: UserWarning: Usage of dash-separated 'license-file' will not be supported in future versions. Please use the underscore name 'license_file' instead
    warnings.warn(
  /home/cim40/anaconda3/lib/python3.8/site-packages/setuptools/dist.py:642: UserWarning: Usage of dash-separated 'license-file' will not be supported in future versions. Please use the underscore name 'license_file' instead
    warnings.warn(
  running bdist_wheel
  running build
  running build_py
  running create_version
  running cmake_build
  -- Build type not set - defaulting to Release
  -- The C compiler identification is GNU 9.3.0
  -- The CXX compiler identification is GNU 9.3.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found Protobuf: /home/cim40/anaconda3/lib/libprotobuf.so (found version "3.14.0")
  CMake Warning at CMakeLists.txt:356 (find_package):
    By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
    asked CMake to find a package configuration file provided by "pybind11",
    but CMake did not find one.
  
    Could not find a package configuration file provided by "pybind11"
    (requested version 2.2) with any of the following names:
  
      pybind11Config.cmake
      pybind11-config.cmake
  
    Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
    "pybind11_DIR" to a directory containing one of the above files.  If
    "pybind11" provides a separate development package or SDK, be sure it has
    been installed.
  
  
  --
  -- ******** Summary ********
  --   CMake version         : 3.18.4
  --   CMake command         : /usr/local/lib/python3.8/dist-packages/cmake/data/bin/cmake
  --   System                : Linux
  --   C++ compiler          : /usr/bin/c++
  --   C++ compiler version  : 9.3.0
  --   CXX flags             :  -Wnon-virtual-dtor
  --   Build type            : Release
  --   Compile definitions   :
  --   CMAKE_PREFIX_PATH     :
  --   CMAKE_INSTALL_PREFIX  : /usr/local
  --   CMAKE_MODULE_PATH     :
  --
[... 169 lines cut ...]
        self.distribution.run_command(command)
      File "/home/cim40/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/home/cim40/anaconda3/lib/python3.8/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/home/cim40/anaconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/cim40/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-v1vncfxm/onnx_4b4b637c32964d11afe6a11a8d86a354/setup.py", line 203, in run
        self.run_command('cmake_build')
      File "/home/cim40/anaconda3/lib/python3.8/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/home/cim40/anaconda3/lib/python3.8/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-v1vncfxm/onnx_4b4b637c32964d11afe6a11a8d86a354/setup.py", line 197, in run
        subprocess.check_call(build_args)
      File "/home/cim40/anaconda3/lib/python3.8/subprocess.py", line 364, in check_call
        raise CalledProcessError(retcode, cmd)
[... cut ...] rom setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hjdpm35p/install-record.txt --single-version-externally-managed --compile --install-headers /home/cim40/anaconda3/include/python3.8/onnx Check the logs for full command output.

This is the full output while installing onnx==1.4.1. My python version is 3.8.5. 
I didnt manage to find where the log file is located

gioriog, 2021-04-26:
[log.txt](https://github.com/jkjung-avt/tensorrt_demos/files/6377543/log.txt)

I didnt manage to find the log file so I am reporting all the output after the command: pip3 install onnx==1.4.1
I am currently using python 3.8.5, maybe I have to downgrade it ? If so how and which version ?

jkjung-avt, 2021-04-26:
Python 3.8.5 should work.  You don't need to downgrade it.

I think the problem you should solve first is the following error.  Check if this helps: https://github.com/onnx/onnx/issues/1624#issuecomment-799314072

> CMake Warning at CMakeLists.txt:356 (find_package):
>       By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
>       asked CMake to find a package configuration file provided [...]

gioriog, 2021-04-26:
Running: sudo pip3 install onnx==1.4.1
Requirement already satisfied: onnx==1.4.1 in /usr/local/lib/python3.8/dist-packages (1.4.1)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from onnx==1.4.1) (1.14.0)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from onnx==1.4.1) (1.17.4)
Requirement already satisfied: typing-extensions>=3.6.2.1 in /usr/local/lib/python3.8/dist-packages (from onnx==1.4.1) (3.7.4.3)
Requirement already satisfied: protobuf in /usr/lib/python3/dist-packages (from onnx==1.4.1) (3.6.1)
Requirement already satisfied: typing>=3.6.4 in /usr/local/lib/python3.8/dist-packages (from onnx==1.4.1) (3.7.4.3)

Seems like it is installed but when I open a python shell and type import onnx, it does not find the module. Maybe I should set some environment variable ?

jkjung-avt, 2021-04-26:
It looks like your "onnx==1.4.1" has been installed at "/usr/local/lib/python3.8/dist-packages".  You should be able to verify that with `pip3 show onnx`.

Otherwise, are you running python in a conda environment?  Make sure "/usr/local/lib/python3.8/dist-packages" is in your python module search paths (`sys.path`).

gioriog, 2021-04-26:
I think I have solved the issue. I had to run python3 yolo_to_onnx.py into sudo python environment instead of python3 yolo_to_onnx.py. Thank you al lot for the advices.

closed 2021-04-26 by gioriog

jkjung-avt, 2022-05-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462

NaveenVSeeWise, 2024-04-12:
I have facing the issue with yolo to onnx conversion using yolov3 tiny pre-trained model.
python3 yolo_to_onnx.py -m yolov3-tiny-288
Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 1065, in <module>
    main()
  File "yolo_to_onnx.py", line 1053, in main
    verbose=True)
  File "yolo_to_onnx.py", line 590, in build_onnx_graph
    params)
  File "yolo_to_onnx.py", line 414, in load_resize_scales
    name, TensorProto.FLOAT, shape, data)
  File "/home/nano/.local/lib/python3.6/site-packages/onnx/helper.py", line 254, in make_tensor
    getattr(tensor, field).extend(vals)
TypeError: 1.0 has type numpy.float32, but expected one of: int, long, float