jkjung-avt/tensorrt_demos #399: How to use the USB camera .

Myron1996, 2021-04-26:
Can you please tell me the command for how to use the USB camera as a live feed to do object detection with yolo . 
@jkjung-avt 

Thank you 
I am getting following error 
`WARNING:root:Camera: starting while cap is not opened!
ERROR: failed to open camera!`

jkjung-avt, 2021-04-26:
"--usb 0" (for /dev/video0) or "--usb 1" (for /dev/video1), and so on.

If it doesn't work, try setting `USB_GSTREAMER = False` in the code (using opencv's built-in mechanism to open USB webcams instead of GStreamer pipeline).

https://github.com/jkjung-avt/tensorrt_demos/blob/4b26770dca33cbe613f65e9b64ec3b9c912e2084/utils/camera.py#L22

Myron1996, 2021-04-26:
Do you mean like this ?
`python3 trt_yolo.py --usb 1 --width 1280 --height 720 -m yolov4-416 -c 80`

Myron1996, 2021-04-26:
Getting the same error 

I tried the following 
`python3 trt_yolo.py --usb 0 -m yolov4-416
`

Myron1996, 2021-04-26:
Hi @jkjung-avt . Its all good . I changed the status  to ` USB_GSTREAMER = False `

and now its working. 
:+1: 

Thank you so much for your quick reply. You are awesome.

closed 2021-04-26 by Myron1996