jkjung-avt/tensorrt_demos #400: Makefile:48: recipe for target 'yolo_layer.o' failed

hudfdfdf, 2021-04-27:
Makefile:20: "compute=75" is for GeForce RTX-2080 Ti.  Please make sure CUDA compute is set correctly for your system in the Makefile.
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/include/x86_64-linux-gnu" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
In file included from /usr/include/c++/5/cstdint:35:0,
                 from /usr/include/x86_64-linux-gnu/NvInferRuntimeCommon.h:54,
                 from /usr/include/x86_64-linux-gnu/NvInferRuntime.h:59,
                 from /usr/include/x86_64-linux-gnu/NvInfer.h:53,
                 from yolo_layer.h:9,
                 from yolo_layer.cu:14:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
Makefile:48: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1

ii  graphsurgeon-tf                                              7.0.0-1+cuda10.0                                amd64        GraphSurgeon for TensorRT package
ii  libnvinfer-bin                                               7.0.0-1+cuda10.0                                amd64        TensorRT binaries
ii  libnvinfer-dev                                               7.0.0-1+cuda10.0                                amd64        TensorRT development libraries and headers
ii  libnvinfer-doc                                               7.0.0-1+cuda10.0                                all          TensorRT documentation
ii  libnvinfer-plugin-dev                                        7.0.0-1+cuda10.0                                amd64        TensorRT plugin libraries
ii  libnvinfer-plugin7                                           7.0.0-1+cuda10.0                                amd64        TensorRT plugin libraries
ii  libnvinfer-samples                                           7.0.0-1+cuda10.0                                all          TensorRT samples
ii  libnvinfer5                                                  5.1.5-1+cuda10.0                                amd64        TensorRT runtime libraries
ii  libnvinfer7                                                  7.0.0-1+cuda10.0                                amd64        TensorRT runtime libraries
ii  libnvonnxparsers-dev                                         7.0.0-1+cuda10.0                                amd64        TensorRT ONNX libraries
ii  libnvonnxparsers7                                            7.0.0-1+cuda10.0                                amd64        TensorRT ONNX libraries
ii  libnvparsers-dev                                             7.0.0-1+cuda10.0                                amd64        TensorRT parsers libraries
ii  libnvparsers7                                                7.0.0-1+cuda10.0                                amd64        TensorRT parsers libraries
ii  python3-libnvinfer                                           7.0.0-1+cuda10.0                                amd64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                                       7.0.0-1+cuda10.0                                amd64        Python 3 development package for TensorRT
ii  tensorrt                                                     7.0.0.11-1+cuda10.0                             amd64        Meta package of TensorRT
ii  uff-converter-tf                                             7.0.0-1+cuda10.0                                amd64        UFF converter for TensorRT package

Thanks for open source. I encountered the above problem when compiling on ubuntu16, do you have any suggestions.

jkjung-avt, 2021-04-27:
1. Please `git pull` the latest code from the repo.

2. If `make` still fails with the latest code, try to modify the following line of code in the Makefile.

    https://github.com/jkjung-avt/tensorrt_demos/blob/4b26770dca33cbe613f65e9b64ec3b9c912e2084/plugins/Makefile#L51

    Replace it with the following instead.

    ```
            $(NVCC) --std c++11 -ccbin $(CC) $(INCS) $(NVCCFLAGS) -Xcompiler "-fPIC" -c -o $@ $<
    ```

STAROFWIND, 2021-05-11:
computes: 75
NVCCFLAGS: -gencode arch=compute_75,code=[sm_75,compute_75]
Makefile:37: *** missing separator.  Stop

Hi, I have an error when  I fix base on the option 2. Any suggestion for that ? Thanks !!