jkjung-avt/tensorrt_demos #409: single-class yolov3 inference problem

reachthedawn, 2021-05-18:
Hi, jkjung,
sincerely appreciate for your excellent work.

Several months ago I trained a yolov3-992 model(from [https://github.com/tanluren/yolov3-channel-and-layer-pruning](url) without its pruning or distillation functions so I think it's similar to AlexyAB's, and using [https://github.com/NVIDIA/apex](url)). This model contains two classes(car and person) works well:

[image]

After conversion, the trt model also works well:

[image]

To achieve faster inference, I train a new yolov3-608 model on single-class(car) with nothing else changed.This new model also works well:

[image]

But after conversion, the trt model runs no result. According to #336, I set confidence threshold to 0.01 in `trt_yolo.py`, and every detection confidence is low, here is the result:

[image]

Comparing the two models and processes, there must be something wrong with single-class inference. Then I notice that the output size at the end of the log when running `onnx_to_tensorrt.py`:

```
$ python3 onnx_to_tensorrt.py -m cls2 -v
......
[TensorRT] VERBOSE: Layer(h884cudnn): 106_convolutional, Tactic: -3237051169894153788, 105_convolutional_lrelu[Half(256,124,124)] -> 106_convolutional[Half(21,124,124)]
[TensorRT] VERBOSE: Layer(Reformat): (Unnamed Layer* 248) [PluginV2IOExt] input reformatter 0, Tactic: 0, 106_convolutional[Half(21,124,124)] -> (Unnamed Layer* 248) [PluginV2IOExt] reformatted input 0[Float(21,124,124)]
[TensorRT] VERBOSE: Layer(PluginV2): (Unnamed Layer* 248) [PluginV2IOExt], Tactic: 0, (Unnamed Layer* 248) [PluginV2IOExt] reformatted input 0[Float(21,124,124)] -> (Unnamed Layer* 248) [PluginV2IOExt]_output_0[Float(322896,1,1)]
Completed creating engine.
```

```
$ python3 onnx_to_tensorrt.py -m cls1 -v
......
[TensorRT] VERBOSE: Layer(h884cudnn): 106_convolutional, Tactic: 2105695814191699972, 105_convolutional_lrelu[Half(256,76,76)] -> 106_convolutional[Half(18,76,76)]
[TensorRT] VERBOSE: Layer(Reformat): (Unnamed Layer* 248) [PluginV2IOExt] input reformatter 0, Tactic: 0, 106_convolutional[Half(18,76,76)] -> (Unnamed Layer* 248) [PluginV2IOExt] reformatted input 0[Float(18,76,76)]
[TensorRT] VERBOSE: Layer(PluginV2): (Unnamed Layer* 248) [PluginV2IOExt], Tactic: 0, (Unnamed Layer* 248) [PluginV2IOExt] reformatted input 0[Float(18,76,76)] -> (Unnamed Layer* 248) [PluginV2IOExt]_output_0[Float(121296,1,1)]
Completed creating engine.
```

It seems that `_output_0[Float(322896,1,1)]` and `_output_0[Float(121296,1,1)]` comes from `totalsize = mYoloWidth * mYoloHeight * mNumAnchors * sizeof(Detection) / sizeof(float)` in `yolo_layer.cu`. And input0size should equal to `(mNumClasses + 5) * mNumAnchors * mYoloHeight * mYoloWidth`. Only when `mNumClasses = 1`,does input0size less than `totalsize`(18 < 21).

I'm not sure if it's the cause of this weird single-class inference result because I'm not familiar with NN layers and TensorRT. And  I don't have other clues.

Sorry for my poor English:D

Thank you very much!

jkjung-avt, 2021-05-21:
I'm not sure the problem is really caused by the yolov3 model being trained with only 1 class...

I myself have trained a custom yolov4-tiny model with only 1 class, and have converted it to a TensorRT engine as well.  As far as I could see, mAP of the TensorRT engine is only slightly worse than the original DarkNet yolov4-tiny model.

Could you check whether you get a better result if you use FP32 instead?  (Refer to https://github.com/jkjung-avt/tensorrt_demos/issues/161#issuecomment-662473343.)

reachthedawn, 2021-05-24:
Nothing changed in FP32.

Another test worked out same problem without APEX.

So I have trained a yolov4-tiny model on my dataset with [https://github.com/AlexeyAB/darknet](). Its TRT engine works well!

[image]

Finally, I think the situation of single-class difference is caused by [https://github.com/tanluren/yolov3-channel-and-layer-pruning](). But I have no time to figure out the reason. I decide to use YOLOv4 to continue my work.

Anyway, thanks for your assistance!

closed 2021-05-24 by reachthedawn