jkjung-avt/tensorrt_demos #410: Problem with using own model

PatrikPatrik2, 2021-05-18:
I use the trt_ssd. Previously I made a model with 7 different classes, I modified the build_engine with using 'num_classes': 8; and also modified the trt_ssd with the 7 classes (besides background) cls_dict = {0:'background', .....
This worked fine.

My new model consists of 10 classes (there are 10 classes in the labelmap.pbtxt that I uses during training). To my suprise I could not use 'num_classes': 11; in the build_engine, but if I instead 12  worked.

I modified the  cls_dict = {0:'background', ..... so it contained the 10 classes

When running trt_ssd I get an error message:
  File "trt_ssd.py", line 101, in main
    loop_and_detect(cam, trt_ssd, conf_th=0.3, vis=vis)
  File "trt_ssd.py", line 67, in loop_and_detect
    img = vis.draw_bboxes(img, boxes, confs, clss)
  File "/home/pi/SSD/project/tensorrt_demos-April_2021/utils/visualization.py", line 96, in draw_bboxes
    color = self.colors[cl]
IndexError: list index out of range

I even tried to increase the cls_dict = {0:'background', 1: 'A', 2: 'B', 3: 'C', 4: 'D', 5: 'E', 6: 'F', 7: 'G', 8: 'H', 9: 'I', 10: 'J', 11: 'K', 12: 'L', 13: 'M', 14: 'N', 15: 'O'}
But it gives the same error. It shows one image from the test avi and thereafter it stopps and shows the error messag.

What can cause the error?

jkjung-avt, 2021-05-20:
I have no idea.  Could you use pdb (or just print it) to check what "cl" (class id) number is output by the TensorRT engine?

jkjung-avt, 2021-05-31:
I'm not going to spend more time to investigate this...

closed 2021-05-31 by jkjung-avt