jkjung-avt/tensorrt_demos #411: very slow and too many detections with yolov4-csp & yolov4x-mish

dotmancando, 2021-05-18:
Hi, jkjung,

First off, thank you very much for your awesome work. I've been testing with your demos especially with yolov3 and yolov4 demos. I have good success running the converted models as in [demo5](https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4). 

However I came across a weird behaviour with yolov4-csp and yolov4x-mish. When I was doing the test on Jetson Xavier NX, I expected the fps to be around 13fps for yolov4-csp-512, but the result I got was around 0.5 fps with loads of detections draw on the image. (please see image below)

[image]

Some notes:
* Other models work fine apart from these 2 models. 
* I made sure to build the "yolo_layer" plugin, when I did the test. 
* I have tested a few times both on other Jetson and also on the same Jetson NX but with fresh install, all have similar behaviour. 

Do you have any suggestions?

Thank you very much for you time.

jkjung-avt, 2021-05-20:
Could you provide your custom "yolov4-csp" and "yolov4x-mish" cfg files?  I will take a look.

dotmancando, 2021-05-20:
Thank you for your response, the cfg files are [here](https://drive.google.com/drive/folders/1HzSBXLEFFgVf8T9D1Cxhe9P_GNUHATj8?usp=sharing)

jkjung-avt, 2021-05-23:
Thanks for reporting the issue.  It is a bug introduced by my [52a699](https://github.com/jkjung-avt/tensorrt_demos/commit/52a699dac94041b28074179d093be6e5669c8741) commit.  It has been fixed in [3b91b8](https://github.com/jkjung-avt/tensorrt_demos/commit/3b91b88fe40e8dd182dfa5cfc0da3b51d82657b4).

Please help to verify it by pulling the latest code from the repo.

dotmancando, 2021-05-24:
Thank you for the fix, I can verify that both `v4-csp` and `v4x-mish` are now working.

closed 2021-05-25 by jkjung-avt