jkjung-avt/tensorrt_demos #413: gstreamer videocapture cause “The file contains no playable streams”

mgabell, 2021-05-21:
I get “The file contains no playable streams”?
I close the capture with cap.release

The gstreamer pipe is:
G_STREAM_TO_DISC = v4l2src device=/dev/video0 ! video/x-raw , width=(int){} , height=(int){} , format=(string){} , framerate=(fraction)60/1 ! timeoverlay ! nvvidconv ! video/x-raw(memory:NVMM), width=(int){}, height=(int){}, format=I420 ! tee name=t t. ! queue ! omxh264enc ! h264parse ! qtmux ! filesink location={}/{}.mp4 t. ! queue ! nvvidconv ! video/x-raw, format=BGRx ! videoconvert ! video/x-raw, format=BGR, width=(int){} , height=(int){} ! appsink

The file is created but unplayable in smplayer.
The reader thread:

def open_camera(g_str):    
    return cv2.VideoCapture(g_str, cv2.CAP_GSTREAMER)

def release(self): 
        logging.debug("Camera thread release camera")        
        try:
            self.cap.release()
            logging.debug("Release succeeded")
        except:
            pass
        self.is_opened = False        
        self._stop()

def _stop(self):         
        logging.debug("Camera thread _stop")        
        if self.camera_thread_running:
            logging.debug("stopping read camera image thread")
            self.camera_thread_running = False
            self.frame_capture_thread.join()

Any ideas appriciated. I have heard about -e to write EOS when terminating, but that seems to be related to console and gst-launch. Also Ihave read about callback, but I dont understand that.. Seems overkill somehow for a 
simple filesink....

After more research I understand that videocapture does not have EOS funktion so you need to use videowriter.
That would mean that I need to pass the video-capture from the appsink to and appsrc and put that into videowriter?

mgabell, 2021-05-21:
I modified the code. changed my first pipe to only create a appsink. then created a new appsrc pipe
that should write to avi.file. So in the thread loop reading camera frames I directly afterwards write using
cv2.VideoWriter(gst_stream,GST....)
I cant get a file created with correct eos but its empty. So I tried to rearrange the writer-pipe to another format. I assumed that was the problem. But I think now I have problem with the actual pipe.

Pipe to Write including a trial of formating to I420 from BGR that didnt work:
G_STREAM_WRITER = appsrc ! nvvidconv ! video/x-raw, format=I420, width=(int)1600 , height=(int)1300 ! videoconvert ! x264enc ! avimux ! filesink location={}/{}.avi

Pipe to read and show on screen. But I try to pick up the appsink as appsrc as seen above and write to disk.
G_STREAM_TO_SCREEN = v4l2src device=/dev/video0 ! video/x-raw , width=(int){} , height=(int){} , format=(string){} , framerate=(fraction)60/1 ! timeoverlay ! nvvidconv ! video/x-raw(memory:NVMM), width=(int){}, height=(int){}, format=I420 ! nvvidconv ! video/x-raw, format=BGRx ! videoconvert ! video/x-raw, format=BGR, width=(int){} , height=(int){} ! appsink

Im lost :(

jkjung-avt, 2021-05-22:
> G_STREAM_TO_DISC = v4l2src device=/dev/video0 ! video/x-raw , width=(int){} , height=(int){} , format=(string){} , framerate=(fraction)60/1 ! timeoverlay ! nvvidconv ! video/x-raw(memory:NVMM), [...]

I'm not sure if this GStreamer pipeline would work when passed as an argument to cv2.VideoCapture().  It contains 1 additional sink ("filesink") along with "appsink" which might cause problems.

Instead, you can record USB webcam video on Jetson using the code snippet below.  The code utilizes the hardware H.264 encoder ("omxh264enc") so that it consumes much less CPU.  The output video file is hardcoded as "output.ts".

```
import numpy as np
import cv2

cap = cv2.VideoCapture(0)
_, img = cap.read()
height, width, _ = img.shape
fps = 30
gst_str = ('appsrc ! videoconvert ! omxh264enc ! mpegtsmux ! '
           'filesink location=output.ts')
writer = cv2.VideoWriter(gst_str, cv2.CAP_GSTREAMER, 0, fps, (width, height))

for _ in range(60):  # capture and write 60 frames
    _, img = cap.read()
    writer.write(img)
    print('.', end='', flush=True)
    # Add additional code here

print()
writer.release()
cap.release()
```

jkjung-avt, 2021-05-25:
@mgabell  Could this issue be closed now?

closed 2021-05-31 by jkjung-avt