jkjung-avt/tensorrt_demos #414: make error in plugins

hiperdyne19015, 2021-05-24:
I wanted to build int8 engine for yolov4. So first I followed the task in demo5. There I checked the pycuda, onnx version and did not install them as I had the updated one already installed.
`
numpy==1.19.5
onnx==1.7.0
pycuda==2019.1.2
`
I tried to make in plugins folder but got this error:

> $.../tensorrt_demos/plugins> make
computes: 75
NVCCFLAGS: -gencode arch=compute_75,code=[sm_75,compute_75]
nvcc -ccbin g++ -I"/usr/local/cuda-10.2/include" -I"/usr/include/x86_64-linux-gnu" -I"/usr/local/include" -I"plugin" -gencode arch=compute_75,code=[sm_75,compute_75] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
In file included from /usr/local/cuda-10.2/include/cuda_runtime.h:83,
                 from <command-line>:
/usr/local/cuda-10.2/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~
Makefile:50: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1

I tried to check if cuda is installed properly. It works well.

> $../tensorrt_demos/plugins> which nvcc
/usr/local/cuda/bin/nvcc

> $../tensorrt_demos/plugins> nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Oct_23_19:24:38_PDT_2019
Cuda compilation tools, release 10.2, V10.2.89

> $/.../1_Utilities/deviceQuery > make
/usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I../../common/inc  -m64    -gencode arch=compute_30,code=sm_30 -gencode arch=compute_35,code=sm_35 -gencode arch=compute_37,code=sm_37 -gencode arch=compute_50,code=sm_50 -gencode arch=compute_52,code=sm_52 -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -o deviceQuery.o -c deviceQuery.cpp
/usr/local/cuda-10.2/bin/nvcc -ccbin g++   -m64      -gencode arch=compute_30,code=sm_30 -gencode arch=compute_35,code=sm_35 -gencode arch=compute_37,code=sm_37 -gencode arch=compute_50,code=sm_50 -gencode arch=compute_52,code=sm_52 -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -o deviceQuery deviceQuery.o 
mkdir -p ../../bin/x86_64/linux/release
cp deviceQuery ../../bin/x86_64/linux/release

> $ ~/.../1_Utilities/deviceQuery > ./deviceQuery 
./deviceQuery Starting...
 CUDA Device Query (Runtime API) version (CUDART static linking)
Detected 1 CUDA Capable device(s)
Device 0: "GeForce RTX 2080 Ti"
  CUDA Driver Version / Runtime Version          11.2 / 10.2
  CUDA Capability Major/Minor version number:    7.5
  Total amount of global memory:                 11016 MBytes (11551440896 bytes)
  (68) Multiprocessors, ( 64) CUDA Cores/MP:     4352 CUDA Cores
  GPU Max Clock rate:                            1545 MHz (1.54 GHz)
  Memory Clock rate:                             7000 Mhz
  Memory Bus Width:                              352-bit
  L2 Cache Size:                                 5767168 bytes
  Maximum Texture Dimension Size (x,y,z)         1D=(131072), 2D=(131072, 65536), 3D=(16384, 16384, 16384)
  Maximum Layered 1D Texture Size, (num) layers  1D=(32768), 2048 layers
  Maximum Layered 2D Texture Size, (num) layers  2D=(32768, 32768), 2048 layers
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 65536
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  1024
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (2147483647, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 3 copy engine(s)
  Run time limit on kernels:                     Yes
  Integrated GPU sharing Host Memory:            No
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Disabled
  Device supports Unified Addressing (UVA):      Yes
  Device supports Compute Preemption:            Yes
  Supports Cooperative Kernel Launch:            Yes
  Supports MultiDevice Co-op Kernel Launch:      Yes
  Device PCI Domain ID / Bus ID / location ID:   0 / 1 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >
deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 11.2, CUDA Runtime Version = 10.2, NumDevs = 1
Result = PASS

I am not sure how to remove this gcc error. Help would be appreciated.

TIA.

jkjung-avt, 2021-05-24:
This error comes exactly from what the error message indicates.

> /usr/local/cuda-10.2/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
> 138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!

This CUDA header file ("crt/host_config.h") does not accept gcc version > 8.  (Maybe you are using gcc 9.x?)

Please use an older version of gcc to build the plugin.

hiperdyne19015, 2021-05-25:
I could make the trt file. 
This link https://linuxhint.com/update_alternatives_ubuntu/ helped me choose gcc 8.

My end goal was to get an int8 engine file so that I could run it on deepstream app. But the generated trt file is not working in deepstream. I could test the trt engine in your repository but it is not working in deepstream.

jkjung-avt, 2021-05-25:
The TensorRT engine built with this repo could be used in DeepStream.  You just have to develop the glue code by yourself.  Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/378.

closed 2021-05-26 by jkjung-avt