jkjung-avt/tensorrt_demos #416: Questions regarding changes to yolo plugin

philipp-schmidt, 2021-05-26:
Hi, I'm trying to update the yolo plugin in https://github.com/isarsoft/yolov4-triton-tensorrt to a more recent version with your fixes to batchsize and updates to compatiblity etc.

Could you briefly explain a few changes that you made to the plugin in comparison to the original from tensorrtx?

1. Input for the yolo layer plugin originally was "all output conv" layers, but now seems to be just one conv layer? So for example the full yolov4 network now needs three instances of the plugin instead of one? And also the output is three marked blobs instead of just one, correct?
2. Original implementation had `MAX_OUTPUT_BBOX_COUNT` how did you handle this change regarding output size etc?
3. What's `input_multiplier` for? I found `input_multiplier = w // yolo_whs[i][0]`. So this is just to not pass `input_width` and `input_height`? It could be calculated if the plugin knew those values?
4. You reduced `int mThreadCount = 256;` to `int mThreadCount = 64;`, is there a performance reason for this?

Thanks already for all those great fixes to the plugin, really helpful.
Could you very briefly explain other changes that you made to  the plugin that you think would be significant enough to mention?

jkjung-avt, 2021-05-27:
> 1. Input for the yolo layer plugin originally was "all output conv" layers, but now seems to be just one conv layer? So for example the full yolov4 network now needs three instances of the plugin instead of one? And also the output is three marked blobs instead of just one, correct?

Yes. I would add 2 or 3 (or more) yolo plugins into the network depending on how many output conv layers there are.  The code is here: https://github.com/jkjung-avt/tensorrt_demos/blob/af878962d27af457f886225e5d51ee2f01fa7afb/yolo/plugins.py#L82-L83

> 2. Original implementation had MAX_OUTPUT_BBOX_COUNT how did you handle this change regarding output size etc?

I don't set an upper limit on output bbox count.  All detection boxes with "scores" higher than the threshold would be kept.  They would go through NMS before the final detection results are generated.  The relevant code is here: https://github.com/jkjung-avt/tensorrt_demos/blob/af878962d27af457f886225e5d51ee2f01fa7afb/utils/yolo_with_plugins.py#L100-L101

> 3. What's `input_multiplier` for? I found `input_multiplier = w // yolo_whs[i][0]`. So this is just to not pass `input_width` and `input_height`? It could be calculated if the plugin knew those values?

Yes, I pass the particular information to the plugin as 1 single value instead of 2.  This is to avoid a problem as described here: https://github.com/NVIDIA/TensorRT/issues/238.  TensorRT plugin code seems to handle `pluginField` values incorrectly if there are too many of them.

> 4. You reduced int mThreadCount = 256; to int mThreadCount = 64;, is there a performance reason for this?

Lower-end Jetson SoCs, such as TX2 and Nano, have only 256 GPU cores in total.  I don't want the yolo plugin to occupy all GPU cores in such systems (i.e. trying to keep some GPU cores available for processing other TensorRT OPs/kernels in parallel).  However, based on my own tests, it doesn't seem to make much difference (between 64 and 256).

philipp-schmidt, 2021-05-27:
Thanks for the answers, they make perfect sense.

closed 2021-05-27 by philipp-schmidt

philipp-schmidt, 2021-05-29:
Hi @jkjung-avt 
Have to bother you again, sorry. I only now realised that your plugin was different from the one in tensorrtx from the very beginning in that it does not compute Logistic Activation. Can you confirm that I have to do Logistic Activation in the last Convolutional Layer before the Yolo Layer myself? Is there anything else that I have to apply to the inputs?

jkjung-avt, 2021-05-29:
For "yolov4-tiny" and "yolov4" models, the conv layers proceeding yolo layers are with "linear" activation, e.g. [yolov4.cfg](https://github.com/AlexeyAB/darknet/blob/5853e51d604712918bd2fb23bab0ec82d19a88f9/cfg/yolov4.cfg)

    ```
    [convolutional]
    size=1
    stride=1
    pad=1
    filters=255
    activation=linear
    ```

In this case, my yolo plugin would calculate "sigmoid" on the input values.  Refer to code here: https://github.com/jkjung-avt/tensorrt_demos/blob/58ad1e9c58af2c8f86a3053a388ee0ef5ca286a5/plugins/yolo_layer.cu#L230-L231

In contrast, "yolov4-csp" and "yolov4x-mish" models would have those conv layers with "logistic" activation, e.g. [yolov4x-mish.cfg](https://github.com/AlexeyAB/darknet/blob/5853e51d604712918bd2fb23bab0ec82d19a88f9/cfg/yolov4x-mish.cfg#L1408-L1413)

    ```
    [convolutional]
    size=1
    stride=1
    pad=1
    filters=255
    activation=logistic
    ```

So in this latter case, I don't need to calculate "sigmoid" in the plugin code: https://github.com/jkjung-avt/tensorrt_demos/blob/58ad1e9c58af2c8f86a3053a388ee0ef5ca286a5/plugins/yolo_layer.cu#L288

In case you are interested in reading the code which handles different types of conv activations, check here: https://github.com/jkjung-avt/tensorrt_demos/blob/58ad1e9c58af2c8f86a3053a388ee0ef5ca286a5/yolo/yolo_to_onnx.py#L699

philipp-schmidt, 2021-05-29:
Yes I only now realised that sigmoid and logistic are practically the same thing. I'm using the TensorRT Layer API (as in tensorrtx) but for converting weights I have to use the python [ScaledYolov4](https://github.com/WongKinYiu/ScaledYOLOv4) Repo. And there seem to be minor differences in the cfg files of the Python Implementation and Darknet which is VERY annoying.

Starting with this:
https://github.com/WongKinYiu/ScaledYOLOv4/issues/202#issuecomment-810913378

And also this:
https://github.com/WongKinYiu/ScaledYOLOv4/issues/202#issuecomment-810917873

And furthermore this (route layer versus route_lhalf):
https://github.com/WongKinYiu/ScaledYOLOv4/issues/165

I guess I'll have to check all implementations for differences...

philipp-schmidt, 2021-05-29:
I'm currently trying to implement yolov4-tiny from here: https://github.com/tjuskyzhang/Scaled-YOLOv4-TensorRT/tree/master/yolov4-tiny-tensorrt

But using your plugin. With little success.

[image]

philipp-schmidt, 2021-05-29:
I believe I need to add a Sigmoid Activation function and use new_coords. I will check that out. Thanks for the help jkjung.

philipp-schmidt, 2021-05-29:
Nevermind.... yolov4-tiny does not use anchor 0....
https://github.com/hunglc007/tensorflow-yolov4-tflite/issues/111#issuecomment-650205659

That fixed it...