jkjung-avt/tensorrt_demos #417: Using the yolo plugin with the C++ api

arxaqapi, 2021-05-27:
Hello, first of all thank you for the awesome work that is being done here.

As a beginner with tensorrt I am struggling to deserialize the **yolo4-tiny-416.trt** engine using the C++ api on the jetson nano.

Any help would be appreciated on how to use the yolo plugin with the C++ api.

Here is the code I use to deserialize the engine:
```cpp
static void load_engine(const string &model_path)
{
    ifstream in_model = read_binary_file(model_path);
    if (!in_model.is_open())
    {
        LOG_ERR("Error opening file");
        exit(EXIT_FAILURE);
    }
    stringstream buff;
    buff << in_model .rdbuf();
    string model_sbuffer = buff.str();
    nvinfer1::IRuntime *runtime = nvinfer1::createInferRuntime(gLogger);
    nvinfer1::ICudaEngine *engine = runtime->deserializeCudaEngine(static_cast<const void *>(model_sbuffer.data()), model_sbuffer.size(), nullptr);
    LOG_OK("Finished loading engine");
}
```
And the error I get while compiling the project:
```
INVALID_ARGUMENT: getPluginCreator could not find plugin YoloLayer_TRT version 1
safeDeserializationUtils.cpp (323) - Serialization Error in load: 0 (Cannot deserialize plugin since corresponding IPluginCreator not found in Plugin Registry)
INVALID_STATE: std::exception
INVALID_CONFIG: Deserialize the cuda engine failed.
```

jkjung-avt, 2021-05-27:
> INVALID_ARGUMENT: getPluginCreator could not find plugin YoloLayer_TRT version 1

This could be easily fixed if you link your executable with either "yolo_layer.o" (statically) or "libyolo_layer.so" (dynamically).  You could find those files under the "plugins/" folder after you do `make`.

fmartinmonier, 2021-05-29:
Hello! I encountered the same issue (also with C++ api) and tried to link my executable with libyolo_layer.so in the following way:

`find_library(YOLO_LAYER NAMES yolo_layer HINTS ${MY_DIR}/tensorrt_demos/plugins)`
and
`include directories (${yolo_layer_INCLUDE_DIRS})`
and
`target_link_libraries(yolo_layer)`

The error is still there. I'm pretty new to using Cmake so I'm not sure I linked the executable correctly. Any idea how to do this ?

jkjung-avt, 2021-05-29:
I'm not familiar with cmake, either.  I could help If you use Makefile instead.

fmartinmonier, 2021-05-29:
Ah that's a shame. I'm working with a ROS package build upon a CMakeList so I have to stick with Cmake unfortunately. Thanks a lot for offering to dive in!

@arxaqapi did you also use Makefile?

jkjung-avt, 2021-05-30:
@fmartinmonier  I think it's probably easier for you to statically link "yolo_layer.o" against your executable.

For example,

```
add_executable(foo foo.cpp $ENV{HOME}/project/tensorrt_demos/plugins/yolo_layer.o)
```

fmartinmonier, 2021-05-30:
Got it. Did that and it works perfectly. Thanks !

closed 2021-05-30 by jkjung-avt

arxaqapi, 2021-05-31:
Hello and thank you for your help, it works like expected by statically linking "yolo_layer.o".

@fmartinmonier I'm am also using cmake.