jkjung-avt/tensorrt_demos #418: Buonding box

Ncnguyen-max, 2021-05-29:
Hi, 
 I do this line:
python3 trt_yolo.py --image ${HOME}/Pictures/dog.jpg -m yolov4-416
[image]

But I don't have the bounding box. Pls help me

jkjung-avt, 2021-05-29:
What's your test platform?  Did you see any error/warning when building the TensorRT engine?

Ncnguyen-max, 2021-05-29:
I use yolov4 and i have few error but i solved it. I don't know why i don't have the buonding box

jkjung-avt, 2021-05-29:
Are you running the TensorRT engine on a Jetson or x86?  What's the problem you've encountered and how did you solve it?

Ncnguyen-max, 2021-05-29:
I running it on Jetson nano. 
I think my problem in here:

$ cd ${HOME}/project/tensorrt_demos/plugins
$ make

computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_53,code=[sm_53,compute_53] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
/bin/sh: 1: nvcc: not found
Makefile:37: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 127

jkjung-avt, 2021-05-29:
> /bin/sh: 1: nvcc: not found

"nvcc" should be located in **/usr/local/cuda/bin**.  You could add that into your PATH environment variable.

Ncnguyen-max, 2021-05-30:
That PATH right?

$ export PATH=/usr/local/cuda/bin${PATH:+:${PATH}}
$ export LD_LIBRARY_PATH=/usr/local/cuda/lib64\{LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}

jkjung-avt, 2021-05-30:
Yes.

Ncnguyen-max, 2021-05-30:
I export 2 lines in /usr/local/cuda/bin. So I have this error:
computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
make: Nothing to be done for 'all'.

Ncnguyen-max, 2021-05-30:
I export 2 lines in /usr/local/cuda/bin. And I do:
$ cd ${HOME}/project/tensorrt_demos/plugins
$ make
But I have this error:
computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
make: Nothing to be done for 'all'.

jkjung-avt, 2021-05-30:
That does not look like an error.  Try `make clean`, then `make` again.

Ncnguyen-max, 2021-05-30:
I make clean and make again. And this again
computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.1/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_53,code=[sm_53,compute_53] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
/bin/sh: 1: nvcc: not found
Makefile:37: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 127

jkjung-avt, 2021-05-31:
Please make sure "/usr/local/cuda/bin/nvcc" is present on your Jetson Nano.  For example,

```
$ ls -l /usr/local/cuda/bin/nvcc
-rwxr-xr-x 1 root root 200792 Oct 30  2019 /usr/local/cuda/bin/nvcc
```

Then add "/usr/loca/cuda/bin" to your PATH.

```
$ export PATH=/usr/local/cuda/bin${PATH:+:${PATH}}
```

Finally, do `make`.

```
$ cd ${HOME}/project/tensorrt_demos/plugins
$ make
```

Ncnguyen-max, 2021-05-31:
Here is my terminal:
ncn@ncn-desktop:~$ ls -l /usr/local/cuda/bin/nvcc
-rwxr-xr-x 1 root root 200792 Thg 1 30  2019 /usr/local/cuda/bin/nvcc
ncn@ncn-desktop:~$ export PATH=/usr/local/cuda/bin${PATH:+:${PATH}}
ncn@ncn-desktop:~$ cd tensorrt_demos/plugins
ncn@ncn-desktop:~/tensorrt_demos/plugins$ make
computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.1/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_53,code=[sm_53,compute_53] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda/lib64" -L"/usr/local/TensorRT-7.1.3.1/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lcudart_static -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group
ncn@ncn-desktop:~/tensorrt_demos/plugins$

jkjung-avt, 2021-05-31:
> ncn@ncn-desktop:/tensorrt_demos/plugins$ make
> computes: 53
> NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
> nvcc -ccbin g++ -I"/usr/local/cuda/include" [...]

This looks OK.  You should have a compiled "libyolo_layer.so" now.  And you could follow the rest of the steps to build and test the TensorRT yolo engines.

Ncnguyen-max, 2021-05-31:
Great! It work.
Thanks you so much.

closed 2021-05-31 by jkjung-avt