jkjung-avt/tensorrt_demos #419: run the trt_yolov3.py  error

kaiven11, 2021-05-31:
1. env
   os:ubuntu 18.04
   python: 3.6.9
   
2. change the class num in yolo.py 
   category_num=5 

3. change the class name dict in yolov3_class.py
4. convert the weights to onnx file (success)
5. convert the onnx to trt file (success)
6. execute commad 
   python3_ trt_yolov3.py --model yolov3-416 --rtsp --uri rtsp://admin:abcd1234@192.168.3.11:554/h264/ch1/main/av_stream
   
    the error message :
    trt_outputs = [output.reshape(shape) for output, shape
    ValueError: cannot reshape array of size 5070 into shape (1,255,13,13)

the cfg file content:
`# Based on cfg/yolov3-voc.cfg

[net]
# Testing
# batch=1
# subdivisions=1
# Training
batch=64
subdivisions=16
width=416
height=416
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=400
max_batches=8200
policy=steps
steps=3800
scales=.1

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=32
size=1
stride=1
pad=1
activation=leaky

[... 706 lines cut ...]
activation=leaky

[convolutional]
size=1
stride=1
pad=1
# filters = (num/3) * (5+classes)
filters=30
activation=linear

[yolo]
mask = 0,1,2
anchors = 10,13,  16,30,  33,23,  30,61,  62,45,  59,119,  116,90,  156,198,  373,326
classes=5
num=9
jitter=.3
ignore_thresh = .5
truth_thresh = 1
random=1

`
weights file:
链接：https://pan.baidu.com/s/1LQVdKnSkQQtLsUbJgDPi3w 
提取码：u39i 
i had try many ways  ,but i still can't resolve the error,pls give me some advice.thk you .

jkjung-avt, 2021-06-01:
"trt_yolov3.py" is a rather old version of the code.  It has been replaced by "trt_yolo.py" now.

Please `git pull` the latest code from the repo and follow the steps in README.md to convert your model.

kaiven11, 2021-06-01:
I had git pull the latest code ,and flollow the steps in README.md, when I  try to excute   "python onnx_to_tensorrt.py -m helmet",

the error message output :
    
[image]

how to solove  this? thk you.

kaiven11, 2021-06-01:
the error message:
[message.txt](https://github.com/jkjung-avt/tensorrt_demos/files/6572975/message.txt)

jkjung-avt, 2021-06-01:
My code assumes the conv features maps before yolo layers are:

https://github.com/jkjung-avt/tensorrt_demos/blob/0276e9b3a7e7f69cdb5fa668f27dd3719c2aa68a/yolo/plugins.py#L90

But your custom model is using "// 4" instead of "// 8".  As a quick fix, you could patch the above line of code with the following:

```python
    yolo_whs = [[w // 32, h // 32], [w // 16, h // 16], [w // 4, h // 4]] 
```

kaiven11, 2021-06-01:
Thk you! It works. 
[image]

kaiven11, 2021-06-01:
The FPS is amazing!

[image]

kaiven11, 2021-06-01:
I tried to use the trt file on  camera  detection, But  i faced a new problem,
[image]

I tried to modify  the conf_th and nms_threshold in yolo_with_plugins.py , But it did not work，Why this happens？

jkjung-avt, 2021-06-01:
This happens when your TensorRT inference runs too fast (inference and draw detection boxes on the same image multiple times).

You could use the "--copy_frame" command-line option to get around that.  (FPS would be slightly lower since the image needs to be copied before each inference.)

kaiven11, 2021-06-01:
i had use the  "--copy_frame" option, And now I‘m trying  to detect 18  rtsp urls,  But  i get a problem.

jkjung-avt, 2021-06-02:
@kaiven11  Please debug the code by yourself.  I don't have time to review it.

kaiven11, 2021-06-02:
Thk you! I tried to run the code , And not found the same problem.