jkjung-avt/tensorrt_demos #420: GStreamer - critical

sawk1, 2021-06-01:
Hi, jkjung-avt, i appreciate your great work, but have this problem in the start of the process : 
![image](https://user-images.githubusercontent.com/46693539/120300621-c641fa00-c2d4-11eb-9627-83871ac3d13f.png
[image]
its still working, but every videostream eats more than 60% of the core on my jetson, so its like im doing decoding on the cpu, but i really want to do it on gpu

jkjung-avt, 2021-06-01:
What kind of video source are you using?

sawk1, 2021-06-02:
rtsp from several cameras

jkjung-avt, 2021-06-02:
I have not tested the code in such scenarios (it'd be difficult for me to test/debug).

I suggest you to verify the RTSP streams one by one.  Make sure they could all be decoded by Jetson hardware individually first.

sawk1, 2021-06-02:
im starting each rtsp in tmux session and this problem shows up in every rtsp, both jetson nano and jetson xt2
also there is a difference between jetson nano and jetson xt2, identical code on this machines - working without any problems on nano, but sometimes showing this error on xt2:
`/utils/mtcnn.py, line 54, in crop_img_with_padding
cropped_im = np.zeroes((ch, cw, 3),dtype=np.uint8)+padding]
ValuerError: negative dimensions are not allowed`
so i add np.abs on calculating ch and cw, but now i have new error:
`utils/mtcnn.py, line 69, in crop_img_with_padding
cropped_im[ey1:ey2, ex1:ex2, :] = img[fy1:fy2, fx1:fx2, :]
ValuerError: could not broadcast input array from shape (5,5,3) into shape(10,10,3`

jkjung-avt, 2021-06-02:
> im starting each rtsp in tmux session and this problem shows up in every rtsp, both jetson nano and jetson xt2

The hardware decoder of Jetson Nano and TX2 cannot handle your RTSP video stream?  I don't have any solution to that...

> also there is a difference between jetson nano and jetson xt2, identical code on this machines - working without any problems on nano, but sometimes showing this error on xt2:

This seems to require some debugging work...  Could you print `cx1, cy1, cx2, cy2` values when this happens?

sawk1, 2021-06-07:
1)hardware decoder is working - 
[image]
i read that jetson tx2 can process 20 video flow in fullhd, so i want to reach half of this with your code(mtcnn part)

2) cx1=147 cx2=148 cy1 = 256 cy2 = 257
its look like it send only  a pixel to process in mtcnn, that is strange.
P.S. as i said before, it happening only on TX2, on jetson nano everything work

jkjung-avt, 2021-06-07:
> 1. hardware decoder is working -
>    ...... i want to reach half of this with your code(mtcnn part)

Please work on that by yourself.

> 2. cx1=147 cx2=148 cy1 = 256 cy2 = 257
>    its look like it send only a pixel to process in mtcnn, that is strange.
>    P.S. as i said before, it happening only on TX2, on jetson nano everything work

If the problem persists, I'd suggest to revert the following commit:  (add back the checking of small PNet boxes)

https://github.com/jkjung-avt/tensorrt_demos/commit/9e6d0e1d6eeb0fc75a8995f96f7aa9d4e4d6e892#diff-d6d81664683345ee77f6a4c25ff52490923e2bde6202e37cd6c15a205c340315

closed 2021-06-13 by jkjung-avt