jkjung-avt/tensorrt_demos #421: label issues

Myron1996, 2021-06-01:
Hi @jkjung-avt . I trained the custom YOLOV3 model with 12 classes and ran an inference. My inference video labels are not showing as CLASS 0 , CLASS 1

Can you please tell me why the names of classes are not showing up ? Do I have to update the classes file. 

Thank you. 
[image]

jkjung-avt, 2021-06-01:
Refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/154#issuecomment-663324412

You could replace `cls_dict`:

https://github.com/jkjung-avt/tensorrt_demos/blob/58ad1e9c58af2c8f86a3053a388ee0ef5ca286a5/trt_yolo.py#L94

with your own class name definition, for example:

```
    cls_dict = {0: 'car', 1: 'bicycle', 2: 'pedestrian'}
```

Myron1996, 2021-06-01:
Thank you @jkjung-avt . It worked. :+1: 

One quick question, 
I was wondering if its possible to use only selected classes to be detected by using pre-trained YOLOV4 model ?

like for example , 

I would like to only `CARS` to be detected and ignore rest of the classes. Is it possible using existing YOLOv4 model or do i have to train new custom model ? 

Thank you.

Myron1996, 2021-06-02:
Does that mean I should put only relevant classes in the following line .

cls_dict = {0: 'car', 1: 'bicycle', 2: 'pedestrian'}

Best Regards
Mayur

jkjung-avt, 2021-06-02:
> Does that mean I should put only relevant classes in the following line .

Not exactly.  You should do filtering of `clss` (class id's of the detected objects) after this line of code.

https://github.com/jkjung-avt/tensorrt_demos/blob/58ad1e9c58af2c8f86a3053a388ee0ef5ca286a5/trt_yolo.py#L66

closed 2021-06-07 by jkjung-avt