jkjung-avt/tensorrt_demos #423: ERROR: failed to open the input video file!

Myron1996, 2021-06-02:
```
tdc@tdc:~/Downloads/tensorrt_demos$ sudo python3 trt_yolo_cv.py -v ./videos/c1.mp4 -o ./c1.mp4 -m yolov4-tiny-416
[sudo] password for tdc: 
Opening in BLOCKING MODE
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[ WARN:0] global /tmp/pip-build-mfax1wyq/opencv-python/opencv/modules/videoio/src/cap_gstreamer.cpp (1825) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module qtdemux0 reported: Internal data stream error.
[ WARN:0] global /tmp/pip-build-mfax1wyq/opencv-python/opencv/modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /tmp/pip-build-mfax1wyq/opencv-python/opencv/modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
ERROR: failed to open the input video file!
```
Any advice on this @jkjung-avt 

Thank you.

jkjung-avt, 2021-06-02:
Refer to the source code: https://github.com/jkjung-avt/tensorrt_demos/blob/52a699dac94041b28074179d093be6e5669c8741/trt_yolo_cv.py#L78

This basically means that cv2 fail to open/play the video file ("./videos/c1.mp4"),  Please use another video file for testing.

Myron1996, 2021-06-02:
I actually tried 4 - 5 video files `.mp4 `format and its still the same error.

jkjung-avt, 2021-06-02:
Maybe you should check your opencv library ("cv2") as well.  The problem is that your "cv2" cannot open/play the video files you are using.  I don't think this is a bug in my code...

Myron1996, 2021-06-08:
Ya you are right . Actually it's working on Jetson tx2 but not on Jetson Nano . But I will look into it . thank you so much.

closed 2021-06-08 by Myron1996

hcliu128, 2023-03-03:
Did you dix it yet? I have the same question now. I'm using video (.MOV and .mp4) to detect on my PC, and it works and detect correctly, but not on Jetson Nano. I have same error code as yours:(
Btw, when I setuped the env, I used pip install -r ./requirement.txt but it broke when installing opencv-python, then I install them separately. I have no idea whether thers's any problem.